makerbase-mks/MKS-TinyBee #1: Microstep settings

wirebeard, 2022-01-05:
Can i configure 32,64 or 128 microstep on board for TMC2209 stepper driver?

MKS-hosiey, 2022-01-06:
Yes, push the dial up to high, and push down to low.

mks-viva, 2022-01-11:
TMC2209 microstep has 8, 32, 64, 16
[image]

radioproektor, 2022-01-16:
This is worst possible manual ever. 
Why use order MS2 MS1 in the chart instead of MS1 MS2 in real switches on the board? 
Why don't arrange things based on microstep size? I looked at datasheet and it is impossible to understand how to set microstep for different driver types. 
What is the purpose of switch 3? Should it be switched always ON?
Why don't make simple and user friendly chart specially for each driver model instead confused text explanation? Example:
[image]

ziofel, 2022-01-20:
I just purchased one of these boards. I can not find the manual anywhere. Can someone please give me a link to the manual. Thanks

MS1987, 2022-01-24:
@radioproektor Wow, greate advice! Thank you.

closed 2022-01-24 by MS1987

LouisR21, 2022-03-05:
> Este es el peor manual posible de la historia. ¿Por qué usar el pedido MS2 MS1 en el gráfico en lugar de MS1 MS2 en los interruptores reales de la placa? ¿Por qué no organizar las cosas según el tamaño del micropaso? Miré el manual de la hoja de datos y es imposible entender cómo configurar el micropaso para diferentes tipos de controladores. ¿Cuál es el propósito del interruptor 3? ¿Debe estar siempre encendido? ¿Por qué no hacer un gráfico simple y fácil de usar especialmente para cada modelo de controlador en lugar de una explicación de texto confusa? Ejemplo: [image]

Thank you very much, I was breaking my head with this problem as well as with another problem to compile with the tmc2209 driver that is solved by defining the drivers as tmc2209 standalone. The manual is really rubbish, I should have more information.

HoverClub, 2022-03-15:
I'm not sure this is correct.  The TinyBee DIP switches connect to Vdd when ON, Gnd (via internal pull-down on 2209) when OPEN.  Switch table should look like this:

**MS1  MS2  uSteps**
OFF  OFF      8
ON    OFF     32
OFF   ON      64
ON     ON      16

Note that MS3 switch MUST be left OFF (open circuit) when using 2209 as it's connected to the Tx output on the 2209.

_References:_
TinyBee Schematic : [https://github.com/makerbase-mks/MKS-TinyBee/blob/main/hardware/MKS%20TinyBee%20V1.0_001/MKS%20TinyBee%20V1.0_001%20SCH.pdf](url)
Trinimatic TMS2209 Datasheet: [https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf](url)

Rainer69, 2022-03-16:
MKS tinybee ist the worst mainboard 4 printing ever seen.