makerbase-mks/MKS-TinyBee #4: NEOPIXEL_LED Backlight

drivernn, 2022-01-10:
Hello! The backlight of the MKS MINI 12864 V3.0_002 screen does not work.  
Please tell me what needs to be fixed? 
Thanks!

[image]

drivernn, 2022-01-16:
Am I the only one having problems with NEOPIXEL_LED Backlight? Does everyone else work?

drivernn, 2022-01-17:
New software don't compile with MKS MINI 12864 V3.0 .

[image]

MKS-hosiey, 2022-01-17:
Is your firmware downloaded from our github or from marlin's page?

drivernn, 2022-01-17:
Yes, from here https://github.com/makerbase-mks/MKS-TinyBee/tree/main/firmware .

wirebeard, 2022-01-18:
Same error on firmware from Marlin page https://github.com/MarlinFirmware/Marlin.git
[image]

MKS-hosiey, 2022-01-18:
Please send your email address, I will send you the firmware that I tested and compiled normally

drivernn, 2022-01-18:
My email address  yyyyymail@yandex.ru .  I will be very grateful for a working firmvere.

ellensp, 2022-01-19:
see https://github.com/MarlinFirmware/Marlin/issues/23561  for the not compiling issue

MKS added the file Mrlin/src/HAL/ESP32/u8g_esp32_spi.cpp but it has a bug

The file start with #ifdef MKS_MINI_12864_V3 but this is checked before the Config files are loaded, so it always fails

Change the start to  
#include "../../inc/MarlinConfigPre.h"
#ifdef MKS_MINI_12864_V3

and delete the 2nd #include "../../inc/MarlinConfigPre.h" line lower down

drivernn, 2022-01-19:
Thanks! It compiles, but there are still problems.

[image]

ellensp, 2022-01-19:
@drivernn that is a com port issue on your machine...    "could not open port `COM9`"   
Its windows, you probably need to reboot as windows has a bad habit of locking com ports.

wirebeard, 2022-01-20:
Firmware compile on Windows 10 with PlatformIO (VSC)  and on Debian 11 with PlatformIO Core (CLI)

drivernn, 2022-01-20:
The first version of the firmware, which was here on github, compiles and works without problems. It had only one problem with the lack of a NEOPIXEL_LED control signal. Is it possible to fix just this minor problem without ruining everything else? )

drivernn, 2022-01-20:
p.s. The problem here is not with the com9 port on my computer, but with the PlatformIO configuration file. I know how to fix it. But I would like to see the official fully working firmware for the standard MKS TinyBee + MKS MINI12864 V3 kit. Unfortunately, such firmware does not exist yet.

solawc, 2022-01-24:
The problem described above has been resolved on the newly updated firmware, and you can download the new one to try. In addition, the default download port specified by the previous firmware is COM9. Now this default download port has been removed. You can see the setting under esp32.ini.

drivernn, 2022-01-25:
Thanks! However, in your latest firmware, MKS_MINI_12864_V3 is enabled again, but //#define NEOPIXEL_LED is disabled. How can the backlight of this screen work??? Is it possible to make a fully working firmware to work with this screen, which is sold in the kit?

michaelslobodskih, 2022-02-26:
> Thanks! However, in your latest firmware, MKS_MINI_12864_V3 is enabled again, but //#define NEOPIXEL_LED is disabled. How can the backlight of this screen work??? Is it possible to make a fully working firmware to work with this screen, which is sold in the kit?

Download latest firmware , uncomment MKS_MINI_12864_V3. Everything works fine (after enabing there are menu for backligth configuration).

closed 2022-03-17 by drivernn

drivernn, 2022-03-17:
I ordered and received a new motherboard. All problems have been resolved. Unfortunately, the first faulty motherboard took me a lot of time and effort.