makerbase-mks/MKS-TinyBee #5: WEB connection and settings

wirebeard, 2022-01-20:
When settings up MKS TinyBee V1.0 from [datasheet](https://github.com/makerbase-mks/MKS-TinyBee/blob/main/MKS%20TinyBee%20V1.0%20datesheet.pdf)

_3. Then click to upload the file on the web control interface, the file
download link: https://github.com/luc-github/ESP3DLib_

_then upload file index,html.zip_

there is no file on the link.

The web interface files has it's own repository [ESP3D-WEBUI](https://github.com/luc-github/ESP3D-WEBUI), required file [index.html.gz](https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz) is on link.

After uploading file, all works.

vasiliyz1979, 2022-07-23:
Hello. Please help, after downloading the html file, it wrote a loading error and an empty page. Nothing works out.
[image]

vasiliyz1979, 2022-08-29:
Hello. After successfully connecting to the access point of the motherboard, the browser does not automatically open a window to download the file that controls the printer, but opens an empty window. Can you tell me what the problem is? How to reset to factory settings.

AngelSerra, 2022-10-01:
my board worked fine, until i changed from marlin 0.9 to 2.1.1 bug fix, wifi stopped working, uploading data through usb works, but there is no wifi or ip, when loading the firware it doesn't give errors, no It can't be connected nor is the load via web or ip, there's nothing, the wifi stopped working and the ip I've scanned all the ip's and nothing doesn't exist, the board is dead, only the USB works.

Marcel67921, 2023-05-04:
[image]

romahow, 2025-10-07:
Скиньте прошивку до мкс тінубей