makerbase-mks/MKS-TinyBee #6: Compile error with TMC2209 drivers

wirebeard, 2022-01-20:
Enabling TMC2209 drivers on the controller BOARD_MKS_TINYBEE results in the error

```
Marlin/src/HAL/ESP32/../../inc/../HAL/ESP32/inc/SanityCheck.h:33:4: error: #error "TMC220x Software Serial is not supported on ESP32."
Marlin/src/HAL/ESP32/../../inc/SanityCheck.h:2897:4: error: #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
```

Configuration.h

```
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
```

Configuration_adv.h

```
 #define  X_SLAVE_ADDRESS 0
 #define  Y_SLAVE_ADDRESS 1
 #define  Z_SLAVE_ADDRESS 2
 #define E0_SLAVE_ADDRESS 3
```

MS1987, 2022-01-24:
Hmm, you should define the configuration as:
#define X_DRIVER_TYPE   TMC2209_STANDALONE
#define Y_DRIVER_TYPE   TMC2209_STANDALONE
#define Z_DRIVER_TYPE   TMC2209_STANDALONE
#define E0_DRIVER_TYPE  TMC2209_STANDALONE

MKS TinyBee doesn't support TMC2209 with uart mode, but standalone as there is no more IOs.

wirebeard, 2022-02-23:
Firmware compile with driver type 

`TMC2209_STANDALONE`

closed 2022-02-23 by wirebeard

AapoTahkola, 2022-04-23:
I got the understanding that you should be able to get them to uart mode if you sacrifice usb uart for it. It would be nice if the manufacturer would explain how to do this so that people would not have to reinvent the wheel.

AapoTahkola, 2022-04-23:
Oh yes the rx and tx are shorted so one needs resolder the pins on steppers to point upwards and then connect them all to TXD2 and RXD2 or TXD and RXD. I think they are both hw uarts. Could be worse I guesss.

cristiantrif, 2022-06-02:
any update here? @wirebeard @MS1987 @AapoTahkola @makerbase-mks @mks-viva

cristiantrif, 2022-06-02:
@AapoTahkola  did you solved it?

AapoTahkola, 2022-06-02:
I did remove R9 and R10 which you can find on tinybee schematic near ESP32 connected to TXD2 and RXD2 if you look closely. Take a look at tinybee TOP to find which ones to remove. It might not be neccessary to remove these because I removed them long before I actually got this to work. Resistor that I used that is in the picture is 220 ohms. The thing is each TMC2209 driver module UART pin has 20K pull down to GND so if you just connect 4 together, the resistor you have to use for the one coming from TX2 is crucial. 1K was fine for two drivers but not if I connected third one so I settled for 220 ohms. It helps if you have reasonably large selection of metal film resistors but in a pinch you a few 560 ohm resistors in parallel will do. You cannot omit the resistor and connect TX2 to RX2 and then to drivers, it will not work. Also you cannot set HARDWARE_SERIAL2_TX and HARDWARE_SERIAL2_RX to same pin, it will not work either.
I have not tested this much yet but I was able to set driver current to 100mA with M906 and get a smooth rotation and expected torque. Considering how thin the wires are that I connected to 12v/24v input that is about as much current I can give to the motors voltage drops dramatically.

Configuration.h:
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209

Configuration_adv.h:
  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 1
  #define  Z_SLAVE_ADDRESS 2
  #define E0_SLAVE_ADDRESS 3
  #define TMC_DEBUG
  
pins_MKS_TINYBEE.h:
#undef HARDWARE_SERIAL2_TX
#undef HARDWARE_SERIAL2_RX

#define HARDWARE_SERIAL2_TX                 17
#define HARDWARE_SERIAL2_RX                 16

#define X_HARDWARE_SERIAL  Serial2
#define Y_HARDWARE_SERIAL  Serial2
#define Z_HARDWARE_SERIAL  Serial2
#define E0_HARDWARE_SERIAL  Serial2

// these are the exp pins that use 17 16
#define EXP1_03_PIN                         -1
#define EXP1_05_PIN                         -1

AapoTahkola, 2022-06-02:
https://www.instagram.com/p/CeSU2uvIIlu/

Position of the switches under stepper drivers:
down, down, down
up, down, down
down, up, down
up, up, down
tmc2209 can share 4 steppers per uart so one is left out.

What you want to do is remove the plastic spacer on one side of the stepper drivers and bend the pin. The 220 ohm resistor is inside the yellow shrink wrap seen on the picture. These wires you see going to the bottom of the picture are not actually related to this mod, I am trying to get i2s as well so I did connect some wires directly to the esp32(epoxy is what the blob there is).

cristiantrif, 2022-06-02:
@AapoTahkola thanks a lot

cristiantrif, 2022-06-29:
I have quit on trying at that time. I have an old GEN L with tmc2208 which is working fine for now. I'll let you know when I'll start working on it again.

AapoTahkola, 2022-06-29:
hese kind of resistors can be removed by heating from the side so that you heat both legs at the same time. You generally cannot solder smd back with soldering iron though. Need to do it Louis Rossmann style with heat gun if you want it back...

3DSmitty, 2022-06-30:
@AapoTahkola did you see my post on #11   "After a little digging it looks like UART1 would be connected to GPIO9 and GPIO10 but in the schematic (SHD/SD2 and SWP/SD3) is not connected to anything???" Maybe that will help.

AapoTahkola, 2022-06-30:
Even on Fysetc Spider board i have has no display and no controls. I just cut the cable and add dupont then connect to relay board to drive chamber heater, air filter, case fan etc. I am not sure we are reading same schematic because I cannot find such symbols in the tinybee schematic. all esp32 pins can be configured for hw uart as I have learned. Theoretically I you can find i2c bus working it is possible to get almost infinite amount of io. Marlin might not support such things but this is a generic esp32 board so all the libraries are there. Sorry for going on a tangent here.. hehe

3DSmitty, 2022-06-30:
On the tinybee schematic:
[image]

From the ESP32-WROOM-32U datasheet:

[image]
[image]

AapoTahkola, 2022-07-06:
Okay had a peek of the board with a microscope and U1, U2 and U3 are 74HC595D. U4, U9 and U10 are hc125. All io on exp1 and exp2 are using 5v io so are the tmc2209 drivers. SSD1306 is supported by marlin but it is 3.3v io max so you cannot just connect it to exp1 or exp2 directly. Also tmc2209 datasheet talks about 5v io. Since TDX2 and RXD2 are just about the only 3.3v io easily available I think I will try to use them for i2c instead(2x PCF8575 together with SSD1306 on the same bus) . There was this OLED_PANEL_TINYBOY2 in Configuration.h which once upon a time apparently has been popular. Virtually all esp32 can be used for hw uart and i2c I have been told. esp32 has 3 uart and if one is used for usb there are two left that can be used for tmc2209 uart. So one might just as well use 4 pins from exp1 and exp2 for 2x uart to get all tmc2209's to uart mode. The bottom pins on esp32 are probably used for spi flash inside the metal can. Pogo pins land on them in factory setting to flash the firmware most likely.

3DSmitty, 2022-07-07:
@AapoTahkola Yes you should be able to use UART2 as long as you don't have anything connected to pins 6 (LCD_D5_0) and 8 LCD_D7_0) on EXP1 as they are tied together.

Emirbal23, 2025-11-01:
I connected the pdn uart pin of tmc2209 to the rx pin of mks tinybee (uart2) and the other pdn uart pin to the tx pin with a resistor in the appropriate position on the switches. 3. Off ms1 and ms2 change according to the addressing but there is still no connection. I get an all low error on the M122. Help me Please