makerbase-mks/MKS-TinyBee #7: Thermistor Temperatures are way off.

CuriousTimo, 2022-02-03:
I have replaced an SKR mini v1.1 with the TinyBee.

#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_BED 1

The temperature on the TinyBee is way off. It reads about 73 degrees for both the bed and E0 whereas the SKR board reads about 18 degrees.
This is at ambient temperature with the printer just sitting idle so I can confirm that the temperature is about 18 degrees.

Gcode M149 confirms that the printer is set to Celcius.
The thermistors are standard 100K

CuriousTimo, 2022-02-03:
Found the issue.

In src\HAL\ESP32\HAL.cpp line 237 has been commented out in the MKS repository.
The value has been changed from 3300.0 to 2600.0

I am using the standard  Marlin 2.9.0.3 repo. So after doing the same change my thermistors are showing the correct temps.

233 void HAL_adc_start_conversion(const uint8_t adc_pin) {
234  const adc1_channel_t chan = get_channel(adc_pin);
235 uint32_t mv;
236  esp_adc_cal_get_voltage((adc_channel_t)chan, &characteristics[attenuations[chan]], &mv);
237  // HAL_adc_result = mv * 1023.0 / 3300.0;
238  HAL_adc_result = mv * 1023.0 / 2600.0;

closed 2022-02-03 by CuriousTimo

HoverClub, 2022-03-13:
Looks like this board uses a 2.5V zener diode as the ADC reference voltage (VDDA measured at 2.5/2.51v on the unconnected TH1/2 pins on a couple of boards). Hard to be certain as there doesn't appear to be a parts list available for this board.  Changing the scaling value to 2500.0 instead of 2600.0 seems to give more accurate results as you'd expect.

infoheleno, 2024-06-03:
muito grato: CuriosoTimo
consegui resover o mesmo problema pela sua ajuda!