makerbase-mks/MKS-TinyBee #8: Compiling error

YoshiAye, 2022-02-03:
Anytime I change something in Configuration.h I'm getting:
```
.pio/build/mks_tinybee/src/src/lcd/dogm/u8g_dev_uc1701_mini12864_HAL.cpp.o:(.data.u8g_dev_uc1701_mini12864_HAL_2x_hw_spi+0x8): undefined reference to `u8g_eps_hw_spi_fn(_u8g_t*, unsigned char, unsigned char, void*)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mks_tinybee\firmware.elf] Error 1
```
I'm changing stuff like: LCD, driver type

Norgulin, 2022-02-11:
Same problem with MKS_LCD12864B
[image]

YoshiAye, 2022-02-22:
> Same problem with MKS_LCD12864B [image]

This is email that I got from them: This is because the library of the new version of the marlin firmware may be missing. We will let engineers fix this, and then feedback to marlin official for modification.

So I guess we should open issue on marlin git.

closed 2022-02-28 by YoshiAye