makerbase-mks/MKS-TinyBee #11: Serial TFT or Wifi but not both?

3DSmitty, 2022-02-20:
According to the docs to use a serial tft you must disable wifi? Can you please explain why?

MKS-hosiey, 2022-02-21:
Because if you use a TFT screen, the Gcode processing is performed on the TF screen, that is, the SD card and the TFT screen, and wifi cannot transfer files to the screen.

3DSmitty, 2022-02-21:
What about the H43 will that work? There is no SD on the H43

elsmpaty, 2022-02-25:
if i connect a mks tft screen with wifi on it.will it work?
which wifi should be disabled?the wifi on board or the wifi on tft?

3DSmitty, 2022-03-29:
Good question elsmpaty, that's what I would like to know. After a little digging it looks like UART1 would be connected to GPIO9 and GPIO10 but in the datasheet (SHD/SD2 and SWP/SD3) is not connected to anything??? Anyways I have a board on the way So I will figure out if I can use the H43 TFT LCD and wifi at the same time.

LouisR21, 2022-03-29:
> if i connect a mks tft screen with wifi on it.will it work? which wifi should be disabled?the wifi on board or the wifi on tft?

I managed to use ( esp32 on board off ) esp3d using the esp12 module for tft and the tft 35 v1.0 With bigtreetech software, but they don't work together. i can see the data inside the sd tft and USB tft but it can be printed from esp3d. There is communication, which is good, more research would be missing, perhaps someone has already made it work. Due to other priorities I was unable to move forward on this issue. I will leave some photos

3DSmitty, 2022-03-29:
Hello rsuzano, I am good. I am still waiting for my board to arrive. Like I said above UART1 would be connected to GPIO9 and GPIO10 but in the datasheet (SHD/SD2 and SWP/SD3) is not connected to anything? If indeed these pins are floating and not connected to anything they could be used for a UART interface for just about anything. We will see...

rsuzano, 2022-05-06:
Hi @3DSmitty  your board arrived??
My thermistor short circuited at heatblock and fried D7 diode, it's part of power circuit, everyting works, but the D7 diode have a hole on it ...and I think this diode it's some kind of protection.
I only find this picture with a baddly D7 label reading... but I don't find any equivalent on google
If your board it's arrived ou anyone wich have this board, can  pass to me de label and/or equivalence of D7 Diode??

[image]

LouisR21, 2022-05-06:
That's the only information it says there. C2
[image]

rsuzano, 2022-05-06:
> That's the only information it says there. C2 [image]

Thanks @LouisR21 ... that simbol after C2 it's what?? only the manufacturer logo?

syntax1269, 2022-05-06:
[image]
On mine it's a bit clearer....

rsuzano, 2022-05-06:
Thanks, I found the datasheet, it's a zener diode

3DSmitty, 2022-05-06:
@rsuzano yes I did receive my board but i have like 10 other projects going on right now so I haven't had a chance to touch it. According to the schematic it looks like some sort of power circuit. I am glad you found the datasheet.
[image]

3DSmitty, 2022-05-16:
@rsuzano   So.. I just discovered Marlin does not support software UART on ESP32 which is needed for TMC2209 UART mode. We might have to wait until Marlin supports this on ESP32 before it will work.

rsuzano, 2022-05-16:
> @rsuzano So.. I just discovered Marlin does not support software serial on ESP32 which is needed for TMC2209 UART mode. We might have to wait until Marlin supports this on ESP32 before it will work.

This limitation it's very disappointing... but, the tmc2209 seems to work well in standalone mode.

About my fried board... I just recoverd it in this weekend... I fried out the 3 schottky diodes of the thermal circuit (t0 t1 and heat bed), the zener diode and the R030 resistor (R17) ... 
After that the heat element of heat block short circuited at the throat heatsink and fried the E0 Mosfet... I'm very unlucky... 
Another problem that's I plugged  the "Anet 12864 Full graphics LCD" on the board and after nothing works I dicovored that positive and negative pints of EXP1 pin header is inverted on the MKS TinyBee... And fried 4 ports of esp32... I need to replace the ESP32 mcu of the board.. 
after many work and some workarouds I managed all to work...

My next adventure was to be the UART ... but now... :(

3DSmitty, 2022-05-17:
@rsuzano Wow how did so many things get fried? You can still try to get the USART1 working I'm just not sure how you will be able to configure it in Marlin.

rsuzano, 2022-05-17:
The sum of inattention, fatigue and bad luck....
I plugger on the LCD of the ANET Full graphics that I already had, and the + and - is inverted, it burned 4 GPIO of the esp32 (Rotary Left, Right, Click, Buzzer)...

I went to put the thermal insulation on the hotend and it damaged the thermistor insulation, short circuited with the heatblock and burned the 30R0 resistor, zener diode and the schottky diodes.

Then I changed my direct-drive extruder for a bowden with heatblock e3d volcano, the heating element closed short on the extruder heatsink and burned the HE0 circuit (mosfet, diodes, and I don't know what else) I changed to HE1 until I found out how to fix it ...

flishy2532, 2023-09-12:
> > if i connect a mks tft screen with wifi on it.will it work? which wifi should be disabled?the wifi on board or the wifi on tft?
> 
> I managed to use ( esp32 on board off ) esp3d using the esp12 module for tft and the tft 35 v1.0 With bigtreetech software, but they don't work together. i can see the data inside the sd tft and USB tft. Can't print sd tft from esp3d. There is communication, which is good, more research would be missing, perhaps someone has already made it work. Due to other priorities I was unable to move forward on this issue. I will leave some photos
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]

did you have config file for tinybee ?

flishy2532, 2023-09-12:
> I managed to use ( esp32 on board off ) esp3d using the esp12 module for tft and the tft 35 v1.0 With bigtreetech software, but they don't work together. i can see the data inside the sd tft and USB [...]

hello did you have config file for mks tinybee ? can you sent to me

mauriciokaster, 2024-02-19:
Hello. The good news is web and display can work together. I succeeded to make the ESP3D webUI working alongside a MKS TFT24 display in a TinyBee control board.
The trick is to configure the SERIAL_PORT_3 section in the _configuration.h_ file to link to USART2 of the TinyBee, which is connected to the TFT.
Then, the serial ports scheme become:

- SERIAL_PORT -> USBSERIAL (cable)
- SERIAL_PORT_2 -> Serial2Socket (WiFi)
- SERIAL_PORT_3 -> USART2 (connected to the TFT)

The changes in the main configuration files:

__Configuration.h:__
```
...
#define SERIAL_PORT 0
...
#define SERIAL_PORT_2 -1
#define BAUDRATE_2 115200  //maybe higher baudrates work
...
#define SERIAL_PORT_3 2
#define BAUDRATE_3 115200  //I configured the TFT to 115200 bps
```

__configuration_adv:__
```
#define ESP3D_WIFISUPPORT  // uncomment this line
```

But some modifications must be done in the HAL files in order to enable this feature:

__src\HAL\ESP32HAL.h:__
_(include these lines somewhere around line 63, before "#define CRITICAL_SECTION_START()...")_
```
typedef ForwardSerial1Class< decltype(Serial2) > DefaultSerial3;
extern DefaultSerial3 MSerial2;
#define MYSERIAL3 MSerial2
```

__src\HAL\ESP32\HAL.cpp:__
_(include these lines somewhere around line 51, before section "// Externs")_
```
#if USING_HW_SERIAL2
  DefaultSerial3 MSerial2(false, Serial2);
#endif
```

But care! It is important to note the ESP3D WebUI and the TFT compete to send GCODE commands to Marlin. Then, it is easy to mess up things trying to use both together.

With such configuration, I can control the printer both with the Web interface and the TFT panel (move X, Y, Z axes, toggle fan on/off, select temperatures for nozzle and for heated bed and so on). I was also able to print with the Gcode file stored in the TFT SD card or by sending the file from the Web interface to the microSD connected to TinyBee. It is possible to send some commands from one interface while a print job is running from the other interface, but It happened to me to lock the web interface once due to any competing gcode command.

*Marlin 2.1.1.1*
*ESP3D 2.1.1 Web interface*
*MKS TinyBee Control board*
*MKS TFT24 Display connected to ESP32 USART2*

joaomarcosvilar, 2025-05-11:
Hello! How are you? I'm facing a series of conflicts when I try to make my code when I try to implement TFT35. Could you make your code available to find my errors? @mauriciokaster

DenisMelnik813, 2025-05-21:
TFT35 USART2 does not work how to fix?