makerbase-mks/MKS-TinyBee #13: Modified library source, breaks compiling after a make clean

ellensp, 2022-02-26:
You have modified files in .pio/libdeps/mks_tinybee/ESP3DLib/src/sd_ESP32.cpp and sd_ESP32.h and web_server.cpp 
You added various references to sd_busy_lock

The instant you do a build clean, the source no longer compiles

If you need to modify these file you need clone the library                    ESP3DLib=https://github.com/luc-github/ESP3DLib/archive/master.zip
Make the change and then reference your cloned repository

MKS-hosiey, 2022-03-08:
Please don't clear it, the modifications we made are all in the library, and the library you downloaded cannot be compiled.

ellensp, 2022-03-09:
As I said, you need to make your own fork of the library with these changes, so correct version gets downloaded and installed.