makerbase-mks/MKS-TinyBee #14: stepping-table and microstepping-settings for TMC2209

DaPeace, 2022-03-02:
Hello, could you please add a stepping-table for the settings to your documentation?
In Marlin there is an option for optimized step-generation for those trinamics stepper drivers that is not mentioned in your documentation.
Also setting the right microstepping is unclear to me. A small table what activates spreadcyle and what stealthchop and what microstepping is used would be very good for the customers. Regards, Jan

MKS-hosiey, 2022-03-05:
Hello, this motherboard does not support uart mode

DaPeace, 2022-03-07:
That was not the question. Question was how to setup the tmc2209s steppings in standalone-mode.

LouisR21, 2022-03-07:
> Hello, could you please add a stepping-table for the settings to your documentation? In Marlin there is an option for optimized step-generation for those trinamics stepper drivers that is not [...]

With this I managed to solve. https://github.com/makerbase-mks/MKS-TinyBee/issues/1

DaPeace, 2022-03-08:
LouisR21 you are the best! Thank you. This explanation is much better then the one in the manual!