makerbase-mks/MKS-TinyBee #15: no movement of the 3d touch.

jaeuwi74, 2022-03-06:
at first, 
You could change the pinout which does not correspond on the pdf (inversion of 5v and GND) this is not serious.
no movement of the 3d touch.
3d touch tested with a BTT skr 1.4 turbo and it works perfectly.
I followed all the steps in the manual and it's not the first one I've installed.
I tested with a second 3d touch, the same.
the 3d touch test on power on is good
please help me

MKS-hosiey, 2022-03-08:
Is the probe not retractable? Has your configuration file enabled the servos, can you upload your configuration file?

jaeuwi74, 2022-03-08:
so I removed the LCD screen and used a screen pin  (21). 
I made the changes in the firmware and now  it works. 
there seems to be a problem with pin 2 of the motherboard.
this is very annoying 
I can't use the board with a screen.

kerocate, 2022-04-02:
Really, I have the same problem, I also tested two 3Dtouch, both clones, bought from Aliexpress.

Except when booting, 3Dtouch will move twice by itself. No probes are thrown at any time.

martings266, 2022-04-28:
Hello, I just bought a mks tiny bee board and Im having problem connecting the 3dtouch, apparently when I turn on the printer, the 3dtouch self-test but afterwards when homing the printer, it doesn't drop the needle not for homing z, neither for auto leveling, I connected the servo to the 3dtouch labeled pins,and the two remaining cables(black and white) to the z endstop socket, I already tried inverting the cables but it does not work. Has anyone managed to fix this? I wouldn't like to lose the lcd function, here is a video of how it doesn't work, the interesting thing is that when the printer boots it does work https://share.icloud.com/photos/0eb3u_8K0NiHdMugYMTSInC8w#Kauno_technologijos_universitetas