makerbase-mks/MKS-TinyBee #17: Sudden voltage drop when activating the heated bed.

LouisR21, 2022-03-08:
My setup: bltouch Smart v3, heated bed 12v 310*310mm, mk8 extruder, tmc2209 drivers, tft 35 v1.0 with esp wifi module, mks mosfet for bed, mks ups, power source 12v 30a

The problem: After 3 days configuring, today I arrive at the time of printing but my surprise It was that when you turn on the heating of the bed, a voltage drop occurs which causes the malfunction It was that when you turn on the heating of the bed, a voltage drop occurs which caused the malfunction of the tft, I already tried connecting the bed directly to the tinybee,And also connecting the MOSFET mks module getting the same result. 

I have a mega with ramps 1.6, for testing and the latter work optimally without absurd voltage drops.

In my ignorance I got curious... Can any other pin on the board be configured to be a signal and thus only use the MOSFET (in signal input mode)

DaPeace, 2022-03-08:
You can edit the pins. In "pins"-folder you will find the pins_MKS_TINYBEE.h and there you could switch the pins. If you have only one extruder you could remove the stepper of E1 and use those pins. Switch #define HEATER_0_PIN 145 -> 141 and #define E1_STEP_PIN 141 -> 145 and you should be fine. Or use any other of the output-pins for this.

MKS-hosiey, 2022-03-08:
Heating bed power down? What is the wattage of your switching power supply? Is it powerful enough to heat the hotbed at full power?

LouisR21, 2022-03-08:
> Heating bed power down? What is the wattage of your switching power supply? Is it powerful enough to heat the hotbed at full power?
> 

the switching power supply is 12v and 30a. the bed does not stop heating but leaves the bltouch, the tft without power.
I'm going to try to use another Power supply, but the current one works fine with Ramos 1.6, 1.4, and the same components.

LouisR21, 2022-03-08:
the switching power supply had a capacitor with a broken solder, and coincidentally it did not work with tinybee and yes with the ramps 🤦

closed 2022-03-08 by LouisR21