makerbase-mks/MKS-TinyBee #18: BABYSTEPPING cause crash

jaeuwi74, 2022-03-08:
problem when I use Marlin's Baby stepping that crash the board
panic error

MKS-hosiey, 2022-03-11:
Did your firmware's advanced settings enable #define BABYSTEP_DISPLAY_TOTAL? need to enable it

jaeuwi74, 2022-03-18:
same error with #define BABYSTEP_DISPLAY_TOTAL

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40089105  PS      : 0x00060b34  A0      : 0x800ff45c  A1      : 0x3ffbe6a0
A2      : 0x3ffbf060  A3      : 0x3ffc27bc  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060b21  A7      : 0x00000000  A8      : 0x80089330  A9      : 0x3ffbe6a0
A10     : 0x00000003  A11     : 0x00060b23  A12     : 0x00060b21  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x3ffc0a45  SAR     : 0x00000019  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x400d2209  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40089105
Backtrace: 0x40089105:0x3ffbe6a0 0x400ff459:0x3ffbe6d0 0x400d2e3f:0x3ffbe700 0x400df253:0x3ffbe730 0x400d3695:0x3ffbe750 0x400df889:0x3ffbe770 0x400df8a3:0x3ffbe790 0x40080fc5:0x3ffbe7b0 0x400847ed:0x3ffbe7d0 0x4000bfed:0x3ffb1e60 0x4008932d:0x3ffb1e70 0x400883af:0x3ffb1e90 0x400e4a21:0x3ffb1eb0 0x400e3065:0x3ffb1ed0 0x400da4d5:0x3ffb1ef0 0x400da524:0x3ffb1f30 0x400d3013:0x3ffb1f50 0x400d3111:0x3ffb1f70 0x400d3195:0x3ffb1f90 0x400e4f31:0x3ffb1fb0 0x40088461:0x3ffb1fd0
Core 0 register dump:
PC      : 0x40103306  PS      : 0x00060134  A0      : 0x800e7236  A1      : 0x3ffbbff0
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800e6dfe  A9      : 0x3ffbbfc0
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80088674  A13     : 0x3ffbbef0
A14     : 0x00000000  A15     : 0x3ffbbce0  SAR     : 0x00000000  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Backtrace: 0x40103306:0x3ffbbff0 0x400e7233:0x3ffbc010 0x400899a5:0x3ffbc030 0x40088461:0x3ffbc050
Rebooting...

3d-hctech, 2022-03-19:
The same issue for me, any solutions pls?
When babystep, the system halt immediately! 

#define BABYSTEP_DISPLAY_TOTAL --- Enabled 
Firmware: Marlin 2.0.9.2
Board: TineyBee V1.0
Screen: MKS H43

himeko, 2022-04-17:
same here, any fix yet?

sinenko, 2022-05-06:
I have this problem too. Is somebody fix this problem?

xmastermail, 2022-05-07:
I confirm. There is a problem. We try different options and versions. So far it doesn't work. Either it reboots or it doesn't move. The problem occurs when the board includes a pause between steps
Firmware: Marlin 2.0.9.2, Marlin 2.0.9.3
Board: TinyBee V1.0

cristiantrif, 2022-06-01:
Any update here? @himeko @jaeuwi74 @sinenko @MKS-hosiey @xmastermail

VooDoo2U, 2022-06-02:
same problem here, looking for a fix, someone has good news?

Cyril13700, 2022-10-02:
Hello.
I just bought the tinybee card.
Unable to operate the babystep.
I tried different version of marlin.
A solution please, I've been on it for 2 days.

Thanks

mashuptwice, 2022-11-27:
Also crashes with the latest `bugfix-2.1.x` branch.
I'll open an upstream issue if I find the time for it.

mashuptwice, 2023-01-13:
Okay, that linked change in `stepper.cpp` did not work. I was still getting a core dump. 

What made a difference was setting `#define MINIMUM_STEPPER_PULSE 3`. Babystepping is still not working, but now Marlin isn't instantly crashing as soon as I turn the knob. I'll try further

Anymorf, 2023-04-29:
How is the situation with babystep crasing. 6 month it lay down because Half functional Marlin firmware. Speed controll - do nothing, babystep - crash and reboot board