makerbase-mks/MKS-TinyBee #19: Thermistor accuracy issue.

HoverClub, 2022-03-13:
See https://github.com/makerbase-mks/MKS-TinyBee/issues/7#issue-1123403716

Looks like this board uses a 2.5V zener diode as the ADC reference voltage (VDDA measured at 2.5/2.51v on the unconnected TH1/2 pins on a couple of boards). Hard to be certain as there doesn't appear to be a parts list available for this board.  However, changing the scaling value to 2500.0 instead of 2600.0 (see link) seems to give more accurate results as you'd expect.

DaPeace, 2022-03-16:
@HoverClub is it possible that this accuracy-issue can cause thermal runaways? Before when i enabled my part-fan the temperatur was suddenly floating up and down a lot. It wasnt even possible to print a benchy. After around a half of the boat the printer stopped because of thermal runaways and the temperature-curve was very spicky.

HoverClub, 2022-03-16:
It depends on which Marlin repo you built from.  The official Marlin one uses the "standard" ESP32 3300.0 divisor (https://github.com/MarlinFirmware/Marlin/pull/23903#issue-1169517970 to fix) and will immediately stop due to non-existent thermal runaway, This repo build uses 2600.0 which reads the temperature as higher than it actually is.    I don't know the specification of the components used on the TinyBee circuit so it's difficult to be certain that 2500 is the correct/best value.  

You can measure the actual voltage on an unconnected thermistor pin on your, warmed up, board and use that value (x1000 obv.) if you want better accuracy.

If it's drifting up and down then there is probably a supply/grounding/noise issue to the thermistor (thermistor ground should connect directly to the TinyBee board and nowhere else and the USB power link removed) - both my boards are pretty steady temp-wise.

HoverClub, 2022-03-16:
It's not very accurate at low temps - mine under reads about 8C at room temp - at "working" temps (>180C) it's around 5C under-reading (compared to a thermocouple on the nozzle - which will be hotter on my head type) so not too bad.  Accuracy is more important at work temps than ambient.  I suspect it's an accumulation of tolerances (Zener diode - typically 5%, thermistor divider resistor - 1%?, thermistor - 0.5% linearity, etc.) but I need to look at the ADC code to see if there are any other issues (and, it'd be good if MKS publish a parts list so that component types & tolerances could be ascertained!).

DaPeace, 2022-03-18:
After changing to the new divider i printed a bit but now i have those thermal problems again and dont know where they come from.
The printer trys to stabilize the temp all the time as you can see in the picture. After changing the temp to 200° its heating up and running stable again. This time the cooling fan is disabled so this should not be a reason.

[image]

HoverClub, 2022-03-18:
Looks like a PID issue to me - a slow oscillation.  Have you tried the PID autotune M303 (you should check it if changing the divisor number as it modifies the sensitivity a little).

DaPeace, 2022-03-19:
I tried to autotune again but i still have those problems. Ive also tried to switch the inputs and outputs to the secondary ports and ive changed the cable to the termistor to a shielded one and its still not save to print. Only thing ive not done yet is to install a new termistor but I think i will try that before buying an other mainboard..

HoverClub, 2022-03-19:
I assume the trace shows the initial stabilised heat up on the left?  If so, it's suspicious that the instability only starts when you print AND that the bed temp is also oscillating.  Could it be a power supply issue (supply drooping when the additional  stepper load is applied) or noise from the stepper drives.  Might be worth putting an oscilloscope on an unconnected thermistor pin (they all use the same ref voltage) and watch what the 2.5V reference voltage is doing (oscillation looks slow enough that a multimeter would probably also work).

DaPeace, 2022-03-19:
As you can see in the trace, the printer is running already for 15-20min and then it starts to oscillate and when changing the temperature it also restabilizes again for a while.  I will swap the powersupply to test it out. Maybe that is the reason..

DaPeace, 2022-03-20:
After changing the Powersupply the printer printed for 4 hours without any issue. Maybe the PSU already had any damage.

countcobolt, 2022-05-08:
Hey guys, I appear to be struggling from a similar thing, but on the bed. When I use a value of 2500, my pid remains more stable, but the bed is underreporting the temp. it should be around 22 and only shows 13. The nozzle is more stable than when using 2600 as a value. Is there another way around it? I am using an original CR10 12V psu, which worked fine on my other board. 
Any clues welcome!
Steve

3DSmitty, 2022-06-21:
Same issue here. I am using a AC Mains Power hot bed and a type 11 thermistor. I just built Marlin 2.1 in which you can now define ADC_REFERENCE_VOLTAGE in your pins file. The default is set to 2.5V. As soon as I turn on the bed the printer errors and halts. If I change it to 2.6V it works but the temp bounces plus or minus 5 degrees and does not become stable. I do not have PID auto tune enabled for the bed at this point as I was not sure if it can be done with a SSR and AC bed. All the docs I found do not mention PID auto tune and AC beds. Can it be done?