makerbase-mks/MKS-TinyBee #21: Random Layershift

ich008, 2022-03-29:
Severe layer shifting (6mm in Y direction 3mm in X direction) wehen it occurs it happens mostly in XY direction at the same time. 
Happens randomly, occurred in 2nd layer after 10minutes aswell as in 20th layer after 30minutes. Tried Marlin2.0.9.2 from here aswell as official 2.0.9.3, same issue. (2.0.x bugfix dosn't compile with bltouch enabled...other issue not mentioned yet). The motor currents are sufficient, speeds and acceleration are reduced (35mm/sec). 
With mesh bed leveling enabled diagonal moves at 180mm/s are stuttery, board randomly froze several times during first print didn't occure since then. Maybe broken Board?

MKS-hosiey, 2022-03-30:
The printing speed is 180mm/s? The speed is too fast to handle. It is recommended that the printing speed be within 150mm/s. The firmware you updated is not on our github? Don't use the firmware on marlin github for now, there are still problems

ich008, 2022-03-30:
The test print above was kept at a max of 35mm/s for both travel and print moves. 

I just expected that fast travel moves with bed leveling enabled also work on such a „fast“ processor which they don’t (at least for me). This was a side note separat to the main problem.

ich008, 2022-04-13:
Without mesh-bed leveling (only G28 in start gcode) printing works fine, even on normal speeds eg. 80mm/s.

ich008, 2022-04-13:
I'll close this issue since it was isolated to me. BTW I reinstalled my old board.

closed 2022-04-13 by ich008

LouisR21, 2022-04-17:
@ich008 let's get in touch, I'm presenting the same problem but multiplied to 1000%. It can be discord, WhatsApp, Facebook to see if we can provide a solution and compare our firmware
[image]

ich008, 2022-04-19:
@LouisR21 
Discord is fine

JazzinSA, 2022-10-03:
THIS is definitely an issue. I just started a new thread and also taken it up with the manufacturers, I too has this issue.