makerbase-mks/MKS-TinyBee #22: Encoder of stock Ender3 display with one-ceble connection doesn't work

Chopper455, 2022-04-03:
I've replaced stock board v4.2.2 of Ender 3 with MKS TinyBee.
As a display was used stock display with one cable.
Connected slots EXP1-EXP1 on both display and the board.
In Marlin configuration I tried several display options, but only several provided boot logo and information screen:
```
#define CR10_STOCKDISPLAY
#define ANET_FULL_GRAPHICS_LCD
#define ANET_FULL_GRAPHICS_LCD_ALT_WIRING 
```
The button part of the controller seems to work fine and allows to enter the menu, beeper also does his job.
But when I turn the wheel, the current menu item doesn't change, as if the encoder isn't responding.
Maybe my encoder is dead or there's something with pinout, unfortunately I can't check on another display.

MKS-hosiey, 2022-04-07:
Yes, this motherboard does not support this display, their interfaces are different

closed 2022-04-10 by Chopper455

Wallnutts, 2022-05-12:
I was able to use a stock ender3/cr10 display using 2 wire connection with the encoder wheel + buzzer working by connecting:
Tinybee  <---> Display
EXP1                  EXP1
EXP2                  EXP2

I also uncommented #define CR10_STOCKDISPLAY