makerbase-mks/MKS-TinyBee #23: Could you please provide bom

cxp297, 2022-04-22:
Could you please provide bom

rsuzano, 2022-05-06:
Please, I need to... My thermistor short circuited at heatblock and fried D7 diode, it's part of power circuit, everyting works, but the D7 diode have a hole on it ...and I think this diode it's some kind of protection.
I only find this picture with a baddly D7 label reading... but I don't find any equivalent on google

[image]

jirka-malvin, 2022-09-15:
all BOM are available at github. Just look up.

irazaq, 2022-09-18:
@jirka-malvin that is not helpful, you could at least give a link to the BOM for tinybee. I am also looking for it too,  there is no BOM file in the hardware directory. 
[image]

jirka-malvin, 2022-09-18:
@irazaq , You are right, I was wrong - BOM is only at first sheet and only partly (capacitors). BOM is not complete at all.

SubJORJ, 2023-08-23:
This is a Zener diode of approximately 2.5~2.7V. The accuracy of readings of temperature sensors TH1, TH2, TB depends on it.
When it failed for me, the temperature sensors began to lie by about -40 degrees.