makerbase-mks/MKS-TinyBee #24: Tune Z-offset while printing not working

countcobolt, 2022-05-01:
Hi all,

I have a sort of frankenstein home made machine, based on a CR10.
Using the tinybee MKS with TMC2209 Stand alone and dual extruder. 
Issue: when starting a print, going to tune and set the z-offset does not do anything. I moved it manually with a large range (like 5 mm etc) but nothing changes.

I have attached my configuration files for your reference.
Is this a bug?

Kind regards
[configuration_files.zip](https://github.com/makerbase-mks/MKS-TinyBee/files/8598714/configuration_files.zip)

flishy2532, 2022-05-04:
i think this line 
/**
 * Set Z_IDLE_HEIGHT if the Z-Axis moves on its own when steppers are disabled.
 *  - Use a low value (i.e., Z_MIN_POS) if the nozzle falls down to the bed.
 *  - Use a large value (i.e., Z_MAX_POS) if the bed falls down, away from the nozzle.
 */
//#define Z_IDLE_HEIGHT Z_HOME_POS

//#define Z_HOMING_HEIGHT  4      // (mm) Minimal Z height before homing (G28) for Z clearance above the bed, clamps, ...
                                  // Be sure to have this much clearance over your Z_MAX_POS to prevent grinding.

//#define Z_AFTER_HOMING  10      // (mm) Height to move to after homing Z

Uncomment to Enable  Comment to Disable

countcobolt, 2022-05-07:
Update: Stock marlin 2.0.9.3 and bugfix allow me to do Babystepping, but the temperature reading is way of. Bugfixing simply sees it as 0 while 2.0.9.3 is about 50C too high. Same thermistor definition etc.

flishy2532, 2022-05-07:
did you have link am test on 2.0.9.3 not compile

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

countcobolt, 2022-05-08:
> did you have link am test on 2.0.9.3 not compile รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>
> […](#)

https://github.com/makerbase-mks/MKS-TinyBee/issues/28

JazzinSA, 2022-10-03:
I think there are 2 different issues in this feed. 
The Baby stepping (as per the title) is an issue. I have a similar issue. The downloadable version of Marlin provided here by Makerbase 2.0.9.3. causes Marlin to crash. I have tried various configurations using A2988 drivers on Extruded and Z-axis and TMC2208_StandAlone on X and Y. Though the option is available, as soon as I start changing the height while the printer is printing layer 1, or any other layer for that matter, Marlin halts and reboots. I thus had to disable this option. I would really like to get an expert to help me with this since we are printer manufacturers an would like to offer this option to our clients.

I simply enabled Babystepping in the ADV file. but if I need to upload any other files please let me know. I simply dont know where to look and fiddle anymore

akhteh, 2025-05-10:
> فکر می‌کنم دو مشکل مختلف در این فید وجود دارد. مشکل گام برداشتن کودک (همانطور که در عنوان آمده است) وجود دارد. من هم مشکل مشابهی دارم. نسخه قابل دانلود Marlin که توسط Makerbase 2.0.9.3 ارائه شده است، باعث از کار افتادن Marlin می‌شود. من پیکربندی‌های مختلفی را با استفاده از درایورهای A2988 روی Extruded و Z-axis و TMC2208_StandAlone روی X و Y امتحان کرده‌ام.
> 
> پس از کامپایل، این گزینه در دسترس است، به محض اینکه شروع به تغییر ارتفاع کنم در حالی که چاپگر در حال چاپ لایه ۱ یا هر لایه دیگری است، مارلین متوقف شده و دوباره راه اندازی می‌شود. بنابراین مجبور شدم این گزینه را غیرفعال کنم. واقعاً دوست دارم از یک متخصص کمک بگیرم تا در این مورد به من کمک کند، زیرا ما تولیدکننده چاپگر هستیم و می‌خواهیم این گزینه را به مشتریان خود ارائه دهیم.
> 
> من به سادگی Babystepping را در فایل ADV فعال کردم. اما اگر نیاز به آپلود فایل‌های دیگری دارم، لطفاً به من اطلاع دهید. من دیگر نمی‌دانم کجا را جستجو و دستکاری کنم.

I don't see this as well. 

> I think there are 2 different issues in this feed. The Baby stepping (as per the title) is an issue. I have a similar issue. The downloadable version of Marlin provided here by Makerbase 2.0.9.3. causes Marlin to crash. I have tried various configurations using A2988 drivers on Extruded and Z-axis and TMC2208_StandAlone on X and Y.
> 
> Once compiled, The the option is available, as soon as I start changing the height while the printer is printing layer 1, or any other layer for that matter, Marlin halts and reboots. I thus had to disable this option. I would really like to get an expert to help me with this since we are printer manufacturers an would like to offer this option to our clients.
> 
> I simply enabled Babystepping in the ADV file. but if I need to upload any other files please let me know. I simply dont know where to look and fiddle anymore

I also have this problem, have you found a solution?