makerbase-mks/MKS-TinyBee #25: Potential pins to reuse for a logical high/low output?

E-Leask, 2022-05-02:
Do you have any pins you recommend i could reuse for a logical high/low? 

I'm planning a dual extruder setup and want to produce a logical high when extruder0 is in use and logical low when extruder1 is in use. Correct me if im wrong but from what I see in the schmetics there are no free digital pins available(if I'm just blind then this makes my life easier). 

Current ideas:
- 3d touch pins: Its a servo so I could reuse the signal pin here.
- Endstop pins: Hypothetically free with sensorless homing, but thereresistors/Schottky diode between esp32 and the pinout so this seems sketchy to me.

Any thoughts?

MKS-hosiey, 2022-05-05:
There are no extra idle pins, the idle enstop pin is ok, but the output level will have a voltage drop of about 0.3V

tekikai, 2022-05-30:
Is the MT-DET pin usable? Am thinking of using it for a homebrew microswitch based leveling sensor.

aleung, 2022-06-26:
> Is the MT-DET pin usable? Am thinking of using it for a homebrew microswitch based leveling sensor.

MT-DET is an input pin. Yes it can be used for probe. I set in `Configuration.h` like this:

``` c
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN 35
```

tekikai, 2022-07-23:
@aleung thanks for confirming, will use it that way!

JulioSugaya, 2023-01-23:
Hello All! 
I would like to control a Neopixel LED through the Marlin features, is that possible by using the servo pin (3d touch)? Or is there another to make this work?
Thanks in Advance.