makerbase-mks/MKS-TinyBee #26: MKS TinyBEE AnyCubic Kossel Linear Plus Marlin 2.0.9.2 Bug LayerShift

flishy2532, 2022-05-04:
any one can help am use mks tinybee on kossel linear plus
tmc2209 x4
am test all config in Marlin 2.0.9.2 have bug layershift it jump when print long 15cm + layer have shift any one can help fix this 
am try test close all command 
S_curve 
Breze_support
change jerk it not help

MKS-hosiey, 2022-05-05:
Layer shift? It may be caused by the motor heating, the motor is not very hot? What is the rated current of the motor? Have you changed the drive current?

flishy2532, 2022-05-05:
am change current drive to 0.8a x3 and 0.6a x1 motor is cool not hot

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

flishy2532, 2022-05-05:
รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

MKS-hosiey, 2022-05-05:
Are you using the firmware from our github?

flishy2532, 2022-05-05:
yes this firmware is your our github

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

MKS-hosiey, 2022-05-05:
Is it normal to go home, move, etc.? Every print is shifted at the same position?

flishy2532, 2022-05-05:
> Is it normal to go home, move, etc.? Every print is shifted at the same position?
> 

it normal go to home but when am test print high position the layer is shifted 
any one can help fix this problem layer shift on Kossel Linear Plus

[image]

[image]

[image]

[image]

[image]

flishy2532, 2022-05-06:
it normal go home but when test print high position the Layer is shifted 
Any One Help Fix

MKS-hosiey, 2022-05-06:
Can you post your config file and Gcode?

flishy2532, 2022-05-06:
yesss sec

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

flishy2532, 2022-05-06:
Here This My G-Code And Config
[G Code and Config.zip](https://github.com/makerbase-mks/MKS-TinyBee/files/8637347/G.Code.and.Config.zip)

flishy2532, 2022-05-06:
any one can help fix LayerShift???

Rainer69, 2022-05-06:
MKS TinyBee is the worst board I have ever bought. After many attempts and unsuccessful solutions, I scrapped the board. I am now using my SKR 1.4 again on my BLV MGN Cube. at least that works

flishy2532, 2022-05-06:
> MKS TinyBee is the worst board I have ever bought. After many attempts and unsuccessful solutions, I scrapped the board. I am now using my SKR 1.4 again on my BLV MGN Cube. at least that works

ah it new board i think wait bug fix

LouisR21, 2022-05-06:
It may be new, but it is a controller board, at least its basic function of completing a print should be 100%. Or announce in your videos that the board is not working properly.
I've tried everything to make it work fine, motor voltage, cooling, don't use hot bed, use other DVR drivers, don't use tft, disable expert functions, I always get the error . It is a prejudice for people who want to start the world of printing with this board, for example my case was like this, I built my own printer from scratch to start thank god I also bought a Ramps with Arduino Mega. I had the same experience with the tft mks 35 v1, a product with flaws, and not enough support. In the end I managed to use it comfortably with  Firmware from bigtreetech.

flishy2532, 2022-05-06:
am delete this bug FI command lol many it have 300+ line omg

[image]

[image]

MKS-hosiey, 2022-05-09:
Hello, is the configuration you sent me the one you are using on your machine? The probe pin conflicts with the z_max pin and it cannot go home. The fan pin in the advanced configuration file also conflicts with the o in the pin.h file, which causes the screen to restart continuously. Are you sure the configuration you sent me is exactly the same as the one used? of?

flishy2532, 2022-05-09:
yes am change probe pin in pin config.h to 35 am see mks tiny bee manual use this pin for mt_det am use this port for probe v2 use micro switch for calibrate menu am mapper new ah delta machine is use max pin for home switch it normal
i dont know what problem why it layer shift when am print around 70-80% on cura

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

MKS-hosiey, 2022-05-09:
I'm verifying this problem in the print test, your model is very slow to print, it takes some time to print to 75%

MKS-hosiey, 2022-05-09:
How many nozzle holes does your machine have? My machine is 0.4, printing your model will extrude too much

flishy2532, 2022-05-09:
yes it 0.4 nozzle am use Fysetc extruder it have gear hahaha

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

MKS-hosiey, 2022-05-09:
Strange, print with your gcode file, the extrusion will be more extra, it looks rough, the parameters of my machine are ok

flishy2532, 2022-05-09:
you can drop rate in step/mm on machine controler my extruder use 450 mm normal is 96

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

MKS-hosiey, 2022-05-11:
[image]

MKS-hosiey, 2022-05-11:
I used your configuration file to print, there is no layer shift problem, you can print my file to test

YoshiAye, 2022-05-13:
> Why am use my config have layer shift @_@ what problem on this am use mks tmc2209 x4 v1.0 Can you help me fix this problem? What am miss. ? Sent from Mail<https://go.microsoft.com/fwlink/?LinkId=550986> for Windows
> […](#)

Your acceleration and current for axes the same as it was on stock board?

flishy2532, 2022-05-13:
yes am set it same

LouisR21, 2022-05-13:
@flishy2532 Your setting is at 1/16 ustep? Today I am going to recompile to test, in my case I had set it to 1/64 and the steps were also calculated at 1/64 . Due to another topic that I read, I am thinking that my problem can be eradicated from there since when interpolating it may be collapsing tinybee (everything is assuming). At first my faults were like yours after playing with speeds and current I came to that mess in the photo.

flishy2532, 2022-05-13:
yes i think it have some bug am alway layer shift how fix them

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

flishy2532, 2022-05-22:
now am fix this problem successful
am test all function and found problem yey 

[image]

[image]

[image]

this fix

[image]

[image]

MKS TMC2209 v1.0

[image]

[image]

[image]

[image]

micro step set

[image]

marlin 2.0.9.2

[image]

sdcard

[image]

[image]

sdcard test[image]

LCD MKS V 3.0 Support RBG Full Color

[image]

[image]

[image]

[image]

[image]

am post this firmware fix soon 👨‍🔧😉

aleung, 2022-06-26:
@flishy2532 How did you fix it?

flishy2532, 2022-06-26:
it not fix am try test am use lady marlin 2.1.x maybe problem in Idle Stepper Shutdown
it #define DEFAUALT_STEPPER_DEACTIVE_TIME 60

am try set to 0 for No Timeout

am test when go 60 min pass it layershift 🤔

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

flishy2532, 2022-10-11:
yes can bump up speed rate on cura or print speed on machine
this model is canopy for rc plane glider
am set slow because need high resolution
and high detel

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>