makerbase-mks/MKS-TinyBee #29: MKS Tinybee error

flishy2532, 2022-05-08:
any one can help i have this error!

[Uploading w{d6f1619} (238e3d48-abf2-40fc-9fc2-d27b8198a191 id=0x7f0a019b tag=CreateIssueFragment) …]()

flishy2532, 2022-05-08:
[image]

MKS-hosiey, 2022-05-09:
The yellow part is a warning, and the red part is an error. Can you take a screenshot of the red font part?

flishy2532, 2022-05-09:
am try find what wronge in this problem am new in platformio any one can help 
old version am use arduino ide it no problem with old motherboard compile but old motherboard is broke (Trigorilla i think original)
[image]

am change to this mks tiny bee
but it layer shift on height print at 73% on cura i dont know what problem on config am use tmc2209 x4 v1.0 on this mks tiny bee
sdcard work
wifi work - esp32 wifi 
screen work - mks_mini v3 with neopixel (cycle color start up)

[image]

[image]

any one can help ? and teach me for platformio
 thank you 
best reguard

flishy2532, 2022-05-09:
did you have delta printer on mks tiny bee?

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

closed 2022-05-28 by flishy2532