makerbase-mks/MKS-TinyBee #31: Error while compiling with TMC2209 UART mode

Wallnutts, 2022-05-11:
[image]

I get this error while compiling if i set TMC2209s up in UART mode:

[image]

Compilation only works if i set it up in TMC2209_STANDALONE mode

flishy2532, 2022-05-11:
use standalone

grafalex82, 2022-10-11:
Is there a chance to enable UART support for TMC2209 on this board?

flishy2532, 2022-10-11:
no am use tmc2209 stand alone it work

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

jirka-malvin, 2022-10-11:
yes UART is possible, but with complications and with no LCD. Already solved in #6

Rmdhn1, 2023-03-11:
> yes UART is possible, but with complications and with no LCD. Already solved in #6

AFAIK you can still use LCD and not many complications, just use Hardware UART.