makerbase-mks/MKS-TinyBee #33: Correct DIP-Settings for TMC2100

tekikai, 2022-05-26:
I am using a little bit older TMC2100 drivers on my Tinybee. 
I am wondering what the correct DIP switch settings would be. 
According to this [https://www.reprap.org/wiki/TMC2100](url) wiki entry I need at least one connection **open** to have 16 microsteps with interpolation but I guess the DIP switches offer GND and VCC as an option only?
Which DIP settings should I choose if I want the best microstepping with either **StealthChop** or **SpreadCycle** mode?
Cheers,
Teki

flishy2532, 2022-05-27:
Here DIP Switch Setting MKS TINYBEE

[image]

tekikai, 2022-05-30:
Thank you for the table! 
I am looking for the settings for TMC2100 specifically. I would need 'GND' and 'open' for 16/256 steps in spread cycle mode and 'open' on switch 1 and 2 for 16/256 stealth chop. I would like to know if it is possible on the Tinybee to configure it that way.
If I am correct for now 16 steps without interpolation is possible (both switches at VCC = switch in upper position?) but for the other modes I would have to desolder the DIP switches?
Just trying to get the best out of the board :)