makerbase-mks/MKS-TinyBee #36: Hotend error

kldmurshed, 2022-05-30:
Hello,
I'm having an error on the HotEnd. Wires connected fine I just uploaded the code and there was no change since last working code. 
I used it for a couple of times heating it, but now showing and error.
Note: HotBed is not connected yet.

Please Help.

[image]

MKS-hosiey, 2022-06-11:
Is your thermal connected? If the thermal is not connected, it will show an error

boshiue, 2022-07-01:
I'm also have the same problem with my thermal connected.....

n3v1l, 2022-07-05:
> Is your thermal connected? If the thermal is not connected, it will show an error

I have the same error, of course the thermals are on the place there should be, I have done every firmware change, the original 2.0.9.2, the latest 2.0.9.4, the 2.0.x-fix and the 2.1.x-fix, all have the same issue. I have done what @boshiue said, and yes the error dissapear now the temps are ok, but the endstop stop working. Is there any solution to this? I have been regreting of have bought this almost useless board. Really p**s off.

westhedge, 2022-07-06:
I have the same issue. I thought the sensors were bad so I tried another brand board and all the sensors tested ok. This is my second tiny bee board. The first board was DOA. I am never going to buy this brand again. The quality is garbage. I’ve tried their official firmware repository and latest marlin firmware and nothing works. Either it’s another junk board or the design is junk or the firmware settings they provide are incorrect. I don’t know which.

westhedge, 2022-07-06:
Also I can confirm that disconnecting the X/Y/Z limit switches allows the thermistors to work. Even connecting one limit switch, the thermistors stop working. So it appears this is a hardware design issue and the board is useless as you can either have working thermistors and no limit switches or you can have limit switches and no working hotends.

boshiue, 2022-07-07:
Here is my connction,because tinybee "S" need to connect to ground  and  trigger endstop at non-connect (NC)
so try this again... and It limit and thermaistors works on the same time on my ender-3
[image]

westhedge, 2022-07-07:
That works with a mechanical end stop but for optical end stops you need v+ and ground as well as signal. Also I measured voltage and the manual says +5 but I measured 3.3v though that probably doesn’t matter.

westhedge, 2022-07-08:
So... I swapped out the optical endstops for mechanical to test this board further and I got about 5 minutes into my calibration when the board suddenly stopped responding to serial commands. The engineering and quality control on these boards is worse than a that of a Ford Pinto. I've re-flashed the firmware multiple times, tried many different computers, rebooted the board and the computers. I've spent HOURS on this POS board. It is NOT worth my time anymore. I'm throwing it directly in the trash and going to a board that doesn't suck so badly. I was testing a BigTree Octopus on this printer and that board worked flawlessly. I've been coding automation for _many_ decades and the MKS TinyBee is by far one of the worse, if not THE worst boards I've ever had the misfortune of wasting my time on.

n3v1l, 2022-07-08:
I agree with you @westhedge this is a POS by far, i have spent a lot of hours trying different setups, when in other hands the old but goodie ramps works without any hassle, very dissapointing. I would like to have a refund, but I bought it on aliexpress and the time for a dispute is over.