makerbase-mks/MKS-TinyBee #39: New UpDate MKS TINYBEE Marlin 2.1.x Fix Bug

flishy2532, 2022-06-12:
This New Marlin for MKS TINYBEE Am Try TEST And Fix Now Work With TMC2209 v1.0 v2.0 with Out Prolem

https://drive.google.com/file/d/1ps3ZbyJAdb8DaHbaFhM4zL12CeH45mGb/view?usp=sharing  This RAR File

https://drive.google.com/file/d/1mj1bVZ6YBHbY9io2gKf1r2sIBk-BAkh2/view?usp=sharing This EXE File

This FirmWare TEST Full WORK

MKS_TINYBEE  Marlin 2.1.x
-MKS v 3.0 RBG full color support 
-WiFi support SDCard support 
-TMC2209 x4 ( v1.0-v2.0) 
###This Config Fix and TEST By. BIRD Engineer ThaiLand ###
contract us : pig_466@hotmail.com 
Line ID : flishy Thank You 
Hope Have fun And Make Item With 3D Print ^.^  _/\_

wirebeard, 2022-06-12:
Do you using UART for stepper driver?

flishy2532, 2022-06-12:
am use tmc and interpola config

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

flishy2532, 2022-06-21:
am change to 2209standalone

3DSmitty, 2022-06-21:
Also interpolation is enabled by default via the hardware.

flishy2532, 2022-06-23:
how to fix layershift on delta printer ? i have this problem on mks tinybee

3DSmitty, 2022-06-23:
most likely either your vref on the tmc drivers are not set correctly or acceleration in marlin is too high.

flishy2532, 2022-06-23:
am set vref 850 am try drop accel

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

flishy2532, 2022-06-23:
[image]

LayerShift aw

fred4537, 2022-06-25:
Hello 
can you try babystepping without probe with fimware 2.0.9.2
because without probe the babystepping no work with me
thank

flishy2532, 2022-06-25:
ok let me test

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

flishy2532, 2022-06-29:
> Hello 
> can you try babystepping without probe with fimware 2.0.9.2
> because without probe the babystepping no work with me
> thank

v 2.0.9.2 have bug babystepping use other version or download bugfix

fred4537, 2022-06-30:
ok thank you

flishy2532, 2022-06-30:
no problem did you use delta printer?

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

fred4537, 2022-07-03:
no I use Anet A8 .
Marlin bugfix 2.1 is good for you

flishy2532, 2022-07-09:
2.1.x have bug surface move too 🥺

3DSmitty, 2022-08-01:
@flishy2532 Are you able to control your part cooling fan with M106? I am not able to turn on the part cooling fan in Marlin 2.1

flishy2532, 2022-08-01:
> @flishy2532 Are you able to control your part cooling fan with M106? I am not able to turn on the part cooling fan in Marlin 2.1

am use auto fan on cura 5.1.0 turn on fan cooling at layer 2

flishy2532, 2022-08-01:
> @flishy2532 Are you able to control your part cooling fan with M106? I am not able to turn on the part cooling fan in Marlin 2.1

[image]

[image]

[image]

[image]

3DSmitty, 2022-08-01:
@flishy2532  Got it thank you! Had to enable
`#define FAN_SOFT_PWM`

Did you have to change the `#define SOFT_PWM_SCALE 0` settings at all?

flishy2532, 2022-08-01:
it work now?

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

3DSmitty, 2022-08-01:
Yes thank you! Did you have to change the #define SOFT_PWM_SCALE 0 settings at all?

flishy2532, 2022-08-01:
> Yes thank you! Did you have to change the #define SOFT_PWM_SCALE 0 setting at all?

only  #define SOFT_PWM_SCALE 0

flishy2532, 2022-08-01:
am print this on my kossel linear plus
[image]

[image]

[image]

3DSmitty, 2022-08-01:
Nice! I put my Tinybee in a Monoprice MP10.
[image]
!

flishy2532, 2022-08-01:
this fame look like my ender3 pro

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

[IMG_20220801_114141032]<https://user-images.githubusercontent.com/51137582/182191445-a7d74203-5a7b-4c00-bbeb-381691460e37.jpg>
!

3DSmitty, 2022-08-01:
> wow all work now?

Yes so far starting to tune in Marlin.

> this fame look like my ender3 pro

Yes similar frame. Bed is AC Mains powered 300x300.

flishy2532, 2022-08-01:
this can print via wifi on cura 5 mks wifi plug in?

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

3DSmitty, 2022-08-01:
No idea I use Prusa Slicer

fred4537, 2022-08-10:
hello 3DSmitty
  your similar ender is ok with Marlin 2.1
 have  you got  MKS lcd 12864  mini V3 ?
your babystepping is ok and your print layer 0.1 is ok

3DSmitty, 2022-08-11:
@fred4537 Yes I am running Marlin 2.1 on the Tinybee on my Monoprice MP10. Yes it is the MKS MINI12864 V3 LCD. I haven't tried babystepping yet and my layer height for printing is 0.2mm. I have printed a few XYZ cubes so far that came out pretty good, I still need to fine tune the acceleration etc.

fred4537, 2022-08-11:
@3DSmitty you can compile marlin 2.1 with visual studio code now ?
because I don't compile marlin 2.1 .
I can just compile  marlin 2.1.1 , but i have not got the lcd. MKS lcd is good with marlin 9.2 and now is black.

3DSmitty, 2022-08-11:
@fred4537 the MKS MINI12864 V3 LCD is working fine for me. Attached is my Marlin config files maybe that will help
[Configuration.txt](https://github.com/makerbase-mks/MKS-TinyBee/files/9312528/Configuration.txt)
[Configuration_adv.txt](https://github.com/makerbase-mks/MKS-TinyBee/files/9312529/Configuration_adv.txt)
.

fred4537, 2022-08-12:
ok thanks

fred4537, 2022-08-13:
@3DSmitty 
Hello
when I compile marlin 2.1 , it's good but the LCD is black. I see your config.
you have maybe  a file marlin 2.1 for tinybee.
 can you share your full marlin file. 
thanks

flishy2532, 2022-08-13:
> @3DSmitty 
> Hello
> when I compile marlin 2.1 , it's good but the LCD is black. I see your config.
> you have maybe  a file marlin 2.1 for tinybee.
>  can you share your full marlin file. 
> thanks

yes i have both 2.1 and 2.1.x bugfix

flishy2532, 2022-08-13:
this 2.1.x bugfix with lcd mks v3.0

fred4537, 2022-08-13:
yes I believe you. I see your machine 
but I don’t understand why lcd lights up with  MKS tinybee Marlin (version 2.0.9)  , and if I configure marlin 2.1 or marlin  2.1.x bugfix of the web site "https://github.com/MarlinFirmware/Marlin and I define MKS_MINI_12864_V3 the LCD.
the Lcd does not light up

3DSmitty, 2022-08-13:
@fred4537 Those are the only two files from Marlin that I configured. All other files are stock from Marlin. Are you sure you 

`#define NEOPIXEL_LED`

It has to be enabled for MKS MINI12864 V3 LCD

3DSmitty, 2022-08-13:
@flishy2532 Did you forget spring cleaning this year? Your desk would make me insane, haha.

fred4537, 2022-08-13:
thank you 
it was #define NEOPIXEL_LED
MKS tinybee marlin (2.0.9) of the siteweb makerbse is disable and it was ok. STRANGE
When I power now printer 3d where I can config the color /default

3DSmitty, 2022-08-13:
@fred4537 oh good glad you figured it out. Happy Printing!

flishy2532, 2022-10-13:
https://drive.google.com/file/d/19MKl1yQN89sEd2PhazUwl3XpRqco5tuS/view?usp=sharing

this exe file fix layer shift bug  change g-code flover in cura to Rearap is bester dont see layer shift

https://drive.google.com/file/d/16XyWPW_-BK43I3r1XOchSx3ZkwAzy-y7/view?usp=sharing

this exe file fix layer shift bug  change g-code flover in cura to Rearap is bester dont see layer shift

flishy2532, 2022-10-13:
[image]

[image]

[image]

[image]

AngelSerra, 2022-10-20:
I know this post is old, but from the photos you need a hot and clean bed with clean and smooth glass, not what you have there, the problem is that the filament does not stick to the hot bed