makerbase-mks/MKS-TinyBee #45: unable to use RepRap 2004 display module

HiIamChaitanya, 2022-07-11:
When I removed the comment #define REPRAP_DISCOUNT_SMART_CONTROLLER
and I build it is showing me the error 'missing lib'

closed 2022-07-11 by HiIamChaitanya

HiIamChaitanya, 2022-12-10:
I enabled other displays as well thus was having error

jaymcd, 2022-12-10:
Ok it s sounds like there is some pin definition issues or non standard display/cable arrangement.Jay M McDonald

HiIamChaitanya, 2022-12-10:
yep , you have to comment other display

jaymcd, 2022-12-10:
I have the right display identified in Marlin. That is not the issue,Jay M McDonald

HiIamChaitanya, 2022-12-10:
then you might have missing lib or can you share their terminal logs / error

jaymcd, 2023-01-03:
I received my MKS Mini 12864 V3 - it does not work with my Tiny Bee either. Defined in Marlin as the MKS Mini 12864 V3 but does not even light up. 

Do I have a defective Tiny Bee board ? 

Jay