makerbase-mks/MKS-TinyBee #46: USB Doesn't Work

jdgabbard, 2022-07-18:
Well, I doubt this comes as a surprise to MKS, even though they have three versions of the board per the GitHub.  However, I'm completely baffled why such an oversight would have taken place.

So, USB is non-functional on the v1.0 board that I have.  At first, I thought it could be a driver issue.  However, after taking the board and looking it over again, I realize that it uses the CH340G (which I am extremely familiar with on a technical level).  And decided this couldn't be the issue. So knowing one of the biggest issues with this USB Serial Adapter is the lack of an adequate clock, I probed pins 7 & 8 of the IC only to discover absolutely no clock signal...

Thinking this was an issue somewhere on the board, I decided to jump on the repository to look at the schematic to back trace the signal and fix the problem.  But low and behold, three versions of the board available in the GitHut Repository, and not a single one of them has the 12mhz Clock signal.  As a matter of fact, those signals were left completely omitted.  And you even left off the oscillator! But even stranger than this, YOU ACTUALLY SOLDERED ON THE USB INTERFACE IC!!!  Why?  Literally, what purpose did this serve if there isn't even a clock signal to drive the controller????  How did this board even make it through Quality Control??? And yes...it even has a QC sticker on the back of it....

I'm also sending a copy of this post to your Support Email.  So the next question has to be obvious, is MKS going to resolve this problem?  How about the (probably low end) thousands of boards you already shipped? There is absolutely no way you didn't know about this.  This would have been caught as soon as you plugged the boards into a USB port....

In case anyone does need a fix, a standard 5v/3.3v XO Oscillator bodged onto the board for +5v and Gnd, and connected to pin 8 of the CH340G, will fix this issue and give you a serial connection to the TinyBee.

mks-viva, 2022-07-18:
Hi! Can you provide same pictures about your board's USB chip?

Kev-hunt, 2022-08-08:
I have just seen your comments. 
I have just upgraded my 3D printer to a MKS TinyBee with a TFT35 smart controller 
I have spent 6 hours over the last 3 days trying to connect my PC's to my Tinybee via USB as detailed by VS Code, youtube and MKS instructions so that I can update my firmware for my 3D printer. 
The TinyBee looks like a relatively new of the line board with the wifi chip, 
Still need to update the firmware though to activate it.

[image]

Image showing the CH340G chip.

Also VS Code has no support for Tinybee.

What are my options? 
I dont feel confident doing the "bodge" as described by @jdgabbard in case I do something wrong.
I think this is extremely bad of MKS to sell faulty boards if this is the case.

Kev-hunt, 2022-08-10:
https://github.com/jdgabbard  Can you please can you post a schematic of your XO fix.

jirka-malvin, 2022-08-30:
have you ESP32 driver installed?

jirka-malvin, 2022-09-30:
> The CH340 has an INTERNAL resonator to provide the required 12MHz clock... [image]
> 
> ..... a missing external crystal is not your problem....

But this is ESP32, not CH340 if I know well

mikeongithub, 2022-10-01:
The conversation was about no comms on an MKS Tinybee....with a CH340G serial - usb chip. The Tinybee schematic clearly shows X0 and X1 disconnected on the CH340....that is not a problem. Look 4 posts up.