makerbase-mks/MKS-TinyBee #49: A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

verissimoster, 2022-07-29:
Whenever I try to flash the FW I always get the error:

Uploading .pio\build\mks_tinybee\firmware.bin
esptool.py v3.0
Serial port COM5
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
======================================================= [FAILED] Took 274.93 seconds =======================================================

Environment    Status    Duration
-------------  --------  ------------
mks_tinybee    FAILED    00:04:34.931

I saw the Topic "USB Doesn't work", but come on I bought a fully functional board and now I'm the one who needs to repair it?! Waiting for feedback from your customer support

DrainIT, 2022-08-25:
I had the same problem. In the opposite corner to the heaters cable and the stepper drivers -near the thermistor sensors cables- there is a button .
Just start uploading, push the button and DON'T release it untile the "connecting" prompt appears, that should be similar in every uploading program you're using - i was using VS code.
I hope that helped.

jirka-malvin, 2022-08-25:
are you using Mac? if yes, do you have driver for ESP installed?

verissimoster, 2022-10-10:
hello guys, tks for your help, I solve the problem by changing the computer, fortunately I do have another one at home, and worked just fine.

closed 2022-10-10 by verissimoster

Rmdhn1, 2023-03-11:
have this issue, used to work, don't know why it's suddenly like this

Rmdhn1, 2023-03-11:
UPDATE : Accidentally plugged in to USB 2.0,
MUST USE 3.0

ntsub, 2023-05-07:
I have still the same problem, uploading on a 3.0 USB with a relatibe short USB cable.
any other idea?

ntsub, 2023-05-07:
SD CART uploading firware.bin does neither work... :(
Totally lost

guigur, 2024-05-18:
got the error on my macbook but it works on my windows desktop

sebmorales, 2025-02-15:
any other ideas on how to fix this on a mac? I have the same problem. Pressing the button during upload (until "connecting") didn't help.

ntsub, 2025-02-16:
The only good typ was to apply a powered USB. I donnot know if it works or
not, because of lack of time and paralel projects