makerbase-mks/MKS-TinyBee #53: Tinybee USB not working

Kev-hunt, 2022-08-17:
What is the issue with Tinybee USB.
I see from other posts with people with the same problem. The USB appears not to work to the computer and thus I can not flash / program the board.
What is the solution.

vasiliyz1979, 2022-08-30:
Entered in the browser http://192.168.0.1 ?forcefallback=yes and everything turned out moved to the loading screen.

jirka-malvin, 2022-08-30:
do you have proper drivers (for ESP32) installed?

vasiliyz1979, 2022-08-30:
Did you rearrange the jumper on the board for downloading via the USB?

Kev-hunt, 2022-08-30:
I am assuming I have ESP32, my other USB devices all work fine.

Yes the jumper is set according to the instructions.

vasiliyz1979, 2022-08-30:
Try flashing the board through the memory card.

andyrok, 2022-08-30:
> Try flashing the board through the memory card.

I’ve the same problem but isn’ t impossible flashing with the memory card…
😩

jirka-malvin, 2022-08-30:
Need more informations how you are trying to flash. So WINDOWS or MACOS?  Do you have installed proper drivers CP210X from silicone Labs (https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads)? If you try to flash - any error message from Visual studion Code? or "nothing happened" when you plug in into USB ? - meaning no LED on at board...

Kev-hunt, 2022-08-31:
I am using Windows 10, VSC and USB to USB.
Nothing happens, 3D printer (Tinybee) is not seeing PC and PC is not seeing Tinybee.

jirka-malvin, 2022-08-31:
> I am using Windows 10, VSC and USB to USB. Nothing happens, 3D printer (Tinybee) is not seeing PC and PC is not seeing Tinybee.

And do you have drivers from siliconlab installed? Jumper is really reconnected? see video https://www.youtube.com/watch?v=6KCa3XEugMY - 1:54 After plugin jumper, LED in the left corner (red) should be ON.

jirka-malvin, 2022-09-13:
BTW: I upgraded and tested tinyBee with Marlin 2.1.1. and works perfectly.