makerbase-mks/MKS-TinyBee #55: Crash at end of M48 test

jgmbrand, 2022-09-02:
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv: Finished!
Recv: Mean: 0.016000 Min: 0.000 Max: 0.027 Range: 0.027
Recv: Standard Deviation: 0.009165
Recv: 
Recv: X:65.00 Y:100.00 Z:3.88 E:0.00 Count X:5265 Y:8100 Z:1551
Recv: 
Recv: Stack smashing protect failure!
Recv: 
Recv: abort() was called at PC 0x400eac68 on core 1
Recv: 
Recv: Backtrace: 0x4008b848:0x3ffb1e20 0x4008ba79:0x3ffb1e40 0x400eac68:0x3ffb1e60 0x400d5597:0x3ffb1e80 0x400d6a0b:0x3ffb1f30 0x400d6dba:0x3ffb1f50 0x400d8590:0x3ffb1f70 0x400d3449:0x3ffb1f90 0x400e7ea5:0x3ffb1fb0 0x40088461:0x3ffb1fd0

AngelSerra, 2022-10-20:
do you have the TMC2209?

jgmbrand, 2022-10-20:
No, I have the A4988 .

Regards ,
Hans