makerbase-mks/MKS-TinyBee #57: the wifi nor the ip work only the USB port

AngelSerra, 2022-10-01:
my board worked fine, until i changed from marlin 0.9 to 2.1.1 bug fix, wifi stopped working, uploading data through usb works, but there is no wifi or ip, when loading the firware it doesn't give errors, no It can't be connected nor is the load via web or ip, there's nothing, the wifi stopped working and the ip I've scanned all the ip's and nothing doesn't exist, the board is dead, only the USB works.

jirka-malvin, 2022-10-02:
Did you enabled wifi support in configuration? Did you upload index.htlm.gz? If yes, did you created file Configuration_Secure.h with login details at same level as othe conf files?

jirka-malvin, 2022-10-02:
Configuration_adv.h -> #define ESP3D_WIFISUPPORT

AngelSerra, 2022-10-02:
yes, it is activated but it still doesn't work only the usb works the #define esp3d_wifisupport is activated and compiled without errors but I can't see the wifi or the ip 192.168.0.1

jirka-malvin, 2022-10-02:
I have installed marlin 2.1.1. with wifi and no problems. Did you created file Configuration_Secure.h as in manual with parameters and put it at same level as config files ? Configuration_Secure.h -> 
`#define WIFI_SSID "SSIDWIFI", #define WIFI_PWD  "PASSWORD FOR WIFI"`

AngelSerra, 2022-10-02:
yes, the file Configuration_Secure.h is put in place. up my files configuations.
[Configuration.zip](https://github.com/makerbase-mks/MKS-TinyBee/files/9693590/Configuration.zip)

jirka-malvin, 2022-10-02:
And your wifiname and password is correct inside of this file?

AngelSerra, 2022-10-02:
yes is correct,
but I can't find the MARLIN_ESP on the wifi, scanning the wifi network does not appear MARLIN ESP.

AngelSerra, 2022-10-02:
> Sí, es correcto, pero no puedo encontrar el MARLIN_ESP en el wifi, escanear la red wifi no aparece MARLIN ESP, according to the manual.

AngelSerra, 2022-10-02:
the board connects with the pronterface without problems, but I can't find the MARLIN_ESP on the wifi.

jirka-malvin, 2022-10-02:
1) if you use Configuration_Secure.h -> you are connecting to your privat wifi network (like home wifi), not to MARLIN_ESP. 
2) If you want to connect MARLIN_ESP wifi, you do not need Configuration_Secure.h file.
Did you followed this tutorial? https://www.youtube.com/watch?v=6KCa3XEugMY

AngelSerra, 2022-10-02:
it needs a factory configuration, something remains in the memory, at the beginning I did the same thing in the tutorial, it stopped working when I changed the firmware version from 1.9 to 2.1.1 bug fix, it only works through the USB port connects with it pronterface program through the USB port, but neither IP no WIFI work, if I have configured the Configuration_Secure.h with my home wifi but it does not connect either.

AngelSerra, 2022-10-19:
Solution:
1st flash deletion, with the ESPtols tool.
2nd apparently it is not compatible with Marlin 2.1, only with Marlin 2.09.
3º in marlin 2.1 the second serial port does not work with many errors, in marlin 2.1 it cannot be programmed, it goes into error mode with a blue led flashing continuously and it does not work, as shown in the previous post.
4º it was possible to restore the board by deleting the flash memory, and installing Marlin 2.09 originated from this website.

with marlin 2.1.1 continuous appearing errors with continuous blue led flashing

ts Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
start
Brown out Reset
Watchdog Reset
Marlin bugfix-2.1.x
echo: Last Updated: 2022-10-16 | Author: (none, default config)
echo: Compiled: Oct 17 2022
echo: Free Memory: 255420 PlannerBufferBytes: 1664
echo:EEPROM version mismatch (EEPROM=? Marlin=V87)
Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x400e80b0 PS : 0x00060930 A0 : 0x800e87d2 A1 : 0x3ffb1e00
A2 : 0x3ffb1e50 A3 : 0x00000000 A4 : 0x000000ff A5 : 0x000000ff
A6 : 0x3ffc3760 A7 : 0x00000000 A8 : 0x00000000 A9 : 0x3ffb1f40
A10 : 0x00000000 A11 : 0x00000000 A12 : 0x000000ff A13 : 0x000000ff
A14 : 0x3ffc3760 A15 : 0x3ffb0060 SAR : 0x00000000 EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000020 LBEG : 0x4000c46c LEND : 0x4000c477 LCOUNT : 0x00000000

Backtrace: 0x400e80b0:0x3ffb1e00 0x400e87cf:0x3ffb1e20 0x400e959a:0x3ffb1ea0 0x400e9b4b:0x3ffb1ec0 0x400d432a:0x3ffb1ee0 0x400d453b:0x3ffb1f00 0x400d4561:0x3ffb1f20 0x400e5549:0x3ffb1f40 0x400e5ab0:0x3ffb1f60 0x400d33c6:0x3ffb1f80 0x400ee163:0x3ffb1fb0 0x40088c59:0x3ffb1fd0

Rebooting...

solved reprogramming the old version.

AngelSerra, 2022-10-20:
By chance I ran into the problem, By chance I have found the problem, this error:

ts Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
start
Brown out Reset
Watchdog Reset
Marlin bugfix-2.1.x
echo: Last Updated: 2022-10-16 | Author: (none, default config)
echo: Compiled: Oct 17 2022
echo: Free Memory: 255420 PlannerBufferBytes: 1664
echo:EEPROM version mismatch (EEPROM=? Marlin=V87)
Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x400e80b0 PS : 0x00060930 A0 : 0x800e87d2 A1 : 0x3ffb1e00
A2 : 0x3ffb1e50 A3 : 0x00000000 A4 : 0x000000ff A5 : 0x000000ff
A6 : 0x3ffc3760 A7 : 0x00000000 A8 : 0x00000000 A9 : 0x3ffb1f40
A10 : 0x00000000 A11 : 0x00000000 A12 : 0x000000ff A13 : 0x000000ff
A14 : 0x3ffc3760 A15 : 0x3ffb0060 SAR : 0x00000000 EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000020 LBEG : 0x4000c46c LEND : 0x4000c477 LCOUNT : 0x00000000

Backtrace: 0x400e80b0:0x3ffb1e00 0x400e87cf:0x3ffb1e20 0x400e959a:0x3ffb1ea0 0x400e9b4b:0x3ffb1ec0 0x400d432a:0x3ffb1ee0 0x400d453b:0x3ffb1f00 0x400d4561:0x3ffb1f20 0x400e5549:0x3ffb1f40 0x400e5ab0:0x3ffb1f60 0x400d33c6:0x3ffb1f80 0x400ee163:0x3ffb1fb0 0x40088c59:0x3ffb1fd0

Rebooting...

the problem is ta here activating this
#define POWER_LOSS_RECOVERY
leaving it like this, everything works wifi LCD etc.. and with the latest version
//#define POWER_LOSS_RECOVERY