makerbase-mks/MKS-TinyBee #58: Layer Shifting - MOTHERBOARD issue

JazzinSA, 2022-10-03:
I'm not sure if this is the right place to look at hardware issues. I suspect that the motherboard has an issue with printing speeds over ~50mm/s. I get Layer shifting with this motherboard which I cannot solve. We are printer manufacturers and know fairly well why layer shifting normally happens, so, NO the steppers is cool and the drivers is also cool and the current is set correctly. 

Has anyone experienced similar issues with this Motherboard?

JazzinSA, 2022-10-03:
I have looked at the SCH of this board and on MAJOR issue with this board is that the board uses serial to paradelle shift registers to increase the IO's. This normally is not a issue however, the "step" pins to the steps sticks is also multiplexed and this is a concern since those pins toggle at serious speeds especially when printing at high speed. This is a over sight on Makerbase's side.

Having said that. I do see LOTS of warnings when compiling. I'm not sure if this is serious but only @Thinkyhead would be able to look at it and say if it would cause issues or not.

JazzinSA, 2022-10-03:
Not so sure anymore that it is a hardware issue since the Z axis and extruder is not affected.

JazzinSA, 2022-10-05:
Ok. To all that have this problem. The solution is as follows.
I'm not sure if it is in fact a motherboard issue or an Marlin issue but the problem is related to writing the EEPROM. It writes to the EEPROM just fine, however there seems to be a problem writing to the EEPROM while printing. The result is that Printing time log and partially doesnt work. Baby Stepping (there might be other futures as well) seems is also affected. The following has to be disabled in your Config to avoid the updating of the EEPROM during printing.

Printstats: 
// #define PRINTCOUNTER 
Theoretically, the print counter CAN be used in Marlin 2.1xx by setting #define PRINTCOUNTER_SAVE_INTERVAL to 0(zero). I havent tested it since I am unable to get Marlin 2.1xx(bugfix) to compile to correctly. It seems that the heater outputs is not working correctly, an issue that I will open anther Bug report on. But if using the version provided here from Makerbase it has to be disabled. THIS is actually what causes the layer shifting.

Babystepping: (This is not confirmed! but it is a high probability)
I'm sure there was a future somewhere that saves the new offset calibrated by babystepping and my suspicion is that this causes Marlin to crash and restart but its not confirmed with the new Marlin I cannot confirm this.

Caian, 2022-10-11:
I'm also noticing layer shifting after changing the control board of my EasyThreed K7 for TinyBee, even with the features above turned off.

thinkyhead, 2022-10-23:
I notice the TinyBee also has the flag…
```cpp
#define USES_SHARED_SPI                           // SPI is shared by SD card with TMC SPI drivers
```
So SPI operations can potentially affect steppers. The EEPROM code uses the library `<EEPROM.h>` via the ESP32 Arduino API, which I assume is i2c, but we can look closer.

When one device interferes with another, some clues might be gleaned by looking at the output of `M43` with `PINS_DEBUGGING` enabled, looking for timer and pin conflicts. The current `bugfix-2.1.x` has a slightly improved version of `M43`, and it may be filled out more over time to show UARTs and other functions that are handled by libraries outside of Marlin.

Caian, 2022-10-23:
> I notice the TinyBee also has the flag…
> 
> ```c++
> #define USES_SHARED_SPI                           // SPI is shared by SD card with TMC SPI drivers
> ```
> 
> So SPI operations can potentially affect steppers. The EEPROM code uses the library `<EEPROM.h>` via the ESP32 Arduino API, which I assume is i2c, but we can look closer.
> 
> When one device interferes with another, some clues might be gleaned by looking at the output of `M43` with `PINS_DEBUGGING` enabled, looking for timer and pin conflicts. The current `bugfix-2.1.x` has a slightly improved version of `M43`, and it may be filled out more over time to show UARTs and other functions that are handled by libraries outside of Marlin.

I noticed the codebased changed a lot between this repo and mainstream Marlin, is the bugfix branch flashable without any changes other the configuration files?

Alt-Pavlo-Delete, 2023-02-20:
Hi guys, i have the same issue with my tinybee v1.0
First i thought that`s it is TMC2208 fault. But it is not.
I already tried to reduce current on the motors to run drivers cooler, tried to lower microstepping, replaced my PSU to the brand new 24V, swapped drivers between X,Y,Z and E and nothing helps. Layers are shifted randomly on the X and Y axis and always at the same time Z axis and Extruder does not affected. Layers shift every 15-20 layers.
I am running official (from the Marlin site) Marlin 2.1.2.

Caian, 2023-02-20:
> Hi guys, i have the same issue with my tinybee v1.0 First i thought that`s it is TMC2208 fault. But it is not. I already tried to reduce current on the motors to run drivers cooler, tried to lower [...]

I also had to disable Backlash compensation (M425) and increase the voltage to the motors, as my 3D printer came with cheap awful ones that will skip if the voltage is too low.

JazzinSA, 2023-02-21:
> I also had to disable Backlash compensation (M425) and increase the voltage to the motors, as my 3D printer came with cheap awful ones that will skip if the voltage is too low.

Hi Caian
I'm not sure how you increased the VOLTAGE on the steppers. Steppers has a very low voltage coil. They actually works on CURRENT from the drivers. so you have to increase the current since increasing the Voltage will have no effect. However if you changed the pot on the drivers then you changed it at the correct place then you just changed the current not the voltage. but anyhow. I successfully eliminated the layer shifting by disabling the PRINTCOUNTER. (see my post above). I'm now printing very successful at 150mm/s without layer shifting. I dont want to tinker too much since it works but I think acceleration could also be the issue. Currently my setting is at 450mm/s/s but I have tested it at 600 and also works well. My Jerk is set to 4 (or whatever its called in marlin 2).

Alt-Pavlo-Delete, 2023-02-21:
Glad to inform that`s after switching from official Marlin 2.1.2 to the current MKS Marlin version (from this GitHub). **Issue gone.**
Issue was 100% not the motors or drivers issue, because drivers were cold, i swapped them from X to Z and issue was still on X axis. Also motors was powerfull enogh because couldnot be stopped by hand during printing, but layers still shifted. And all this untill i flashed the firmware from the current GitHub repo.

Caian, 2023-02-21:
> > > Hi guys, i have the same issue with my tinybee v1.0 First i thought that`s it is TMC2208 fault. But it is not. I already tried to reduce current on the motors to run drivers cooler, tried to lower microstepping, replaced my PSU to the brand new 24V, swapped drivers between X,Y,Z and E and nothing helps. Layers are shifted randomly on the X and Y axis and always at the same time Z axis and Extruder does not affected. Layers shift every 15-20 layers. I am running official (from the Marlin site) Marlin 2.1.2.
> > 
> > 
> > I also had to disable Backlash compensation (M425) and increase the voltage to the motors, as my 3D printer came with cheap awful ones that will skip if the voltage is too low.
> 
> Hi Caian I'm not sure how you increased the VOLTAGE on the steppers. Steppers has a very low voltage coil. They actually works on CURRENT from the drivers. so you have to increase the current since increasing the Voltage will have no effect. However if you changed the pot on the drivers then you changed it at the correct place then you just changed the current not the voltage. but anyhow. I successfully eliminated the layer shifting by disabling the PRINTCOUNTER. (see my post above). I'm now printing very successful at 150mm/s without layer shifting. I dont want to tinker too much since it works but I think acceleration could also be the issue. Currently my setting is at 450mm/s/s but I have tested it at 600 and also works well. My Jerk is set to 4 (or whatever its called in marlin 2).
> 
> I still have A LOT of warnings when compiling though which does bother me quite a lot and I'm not sure how to fix them. I soooo wish someone with deeper knowledge with the code can help me fix them since they are so deep inside Marlin that I get a bit lost. If I may ask what screen are you running since the more fancy screens also use the SPI and could also cause issues.

Sorry, you are right, I meant the VRef of the stepper driver.

thinkyhead, 2023-07-20:
> Glad to inform that`s after switching from official Marlin 2.1.2 to the current MKS Marlin version (from this GitHub). Issue gone.

We'll have to compare these different versions and see if anything stands out. Was the same exact set of configuration options used with both versions?