makerbase-mks/MKS-TinyBee #59: Information missing from the WebUI

Caian, 2022-10-07:
I've cloned the latest version (5f248e6a6cd968b85ac0a9818a4e4b48c0eca5c2) of this repository, built and flashed the firmware without any issues, managed to connect to the printer and upload the WebUI gz file that is also in this commit.

The printer is printing from the WebUI without issues, however, it is not showing information like the temperature graph, also all commands in the command panel only return "Ok", even those that are supposed to return additional information like `M105`, `M118`,...

How do I fix this issue?

Caian, 2022-10-07:
Already tried, the horizontal lines of the graph show up, but nothing is plotted.

jirka-malvin, 2022-10-07:
[image]

It shoul looks like this pisture ... At me - this is working

Caian, 2022-10-07:
I found the issue, I was tunneling my connection through `ssh` into my network. But apparently ESP32Lib creates a websocket for the WebUI that is not being tunneled.

closed 2022-10-07 by Caian