makerbase-mks/MKS-TinyBee #65: Couldn't connect H43 or TS35

deadstarcan, 2022-11-12:
I couldnt connect or initialize both of these display sets. Of course I will use only one, Do you know any solution?

Moditek, 2022-12-11:
Configuration.h

#define DGUS_LCD_UI_MKS

and

#define SERIAL_PORT 1

#define BAUDRATE 115200

Configuration_adv.h

#if HAS_DGUS_LCD
#define LCD_SERIAL_PORT 2
  #define LCD_BAUDRATE 115200

  #define DGUS_RX_BUFFER_SIZE 128
  #define DGUS_TX_BUFFER_SIZE 48
  //#define SERIAL_STATS_RX_BUFFER_OVERRUNS  // Fix Rx overrun situation (Currently only for AVR)

jorgejglopez, 2023-01-21:
I have the MKS H43 working with a MKS Robin Nano V3.0. This is my configuration for the screen. Also I have the screen connected on the pins marked as (PB10, PB11) on my board (TX3 and RX3).
My screen didn't boot up the first time, and I have to reflash the firmware screen. For me the version that work is the 1.30. For the reflash its very important that you make al the steps described in the GitHub page: [MKS-H43](https://github.com/makerbase-mks/MKS-H43)

Configuration.h
```
#define DGUS_LCD_UI_MKS
#if ENABLED(DGUS_LCD_UI_MKS) 
  //#define USE_MKS_GREEN_UI
#endif

#define SERIAL_PORT 1

#define BAUDRATE 115200
```
Configuration_adv.h
```
#if HAS_DGUS_LCD
  #define LCD_SERIAL_PORT 3
  #define LCD_BAUDRATE 115200

  #define DGUS_RX_BUFFER_SIZE 128
  #define DGUS_TX_BUFFER_SIZE 48
  //#define SERIAL_STATS_RX_BUFFER_OVERRUNS  // Fix Rx overrun situation (Currently only for AVR)

```

sherwinrastegar, 2023-09-11:
Hello, 
how was your problem solved? I have the same problem, and when I compile, it says that the board does not support this display.