makerbase-mks/MKS-TinyBee #75: Error writing sd card using wifi

carlosm180, 2022-12-11:
I try to send a file (a g-code) using web ui, but it is not able to write sd card (trying with differents sd card)....transfert is shown completed (100%), but no files are on the media. Also using Cura5.1 and MKS WIFI plug-in there is nothing to do. Coping files using a card reader, instead, I have no problem. Any idea?

Uka4me, 2022-12-11:
I will join the problem. Tried formatting cards via https://www.sdcard.org/downloads/formatter/, update ESP3DLib, WebUI, marlin to 2.1.1. As a result, when you try to send a file, you get the error "File write failed". This creates the file itself with the ending "S" and has a size of 0 bytes. Delete files, create and delete folders works, but does not upload. I use the author's scheme, but it is extremely inconvenient.

Uka4me, 2022-12-12:
@carlosm180 Can you share your Configuration.h file? The SD_CHECK_AND_RETRY option was initially active for me, but even with reinstallations I can’t succeed like you.

carlosm180, 2022-12-13:
This is my configuration.... it is not perfect - I need to refine it - but it works with ufficial Tinybee Marlin. Please note I have 2 extruders, so I have disabled sdcard reader on board (by jumper), using MKS Mini 12864 v.3 reader.

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 02000902

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some useful links to help get your machine configured and calibrated:
 *
 * Example Configs:     https://github.com/MarlinFirmware/Configurations/branches/all
 *
 * Průša Calculator:    https://blog.prusaprinters.org/calculator_3416/
 *
 * Calibration Guides:  https://reprap.org/wiki/Calibration
 *                      https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 *                      https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 *                      https://youtu.be/wAL9d7FgInk
 *
 * Calibration Objects: https://www.thingiverse.com/thing:5573
 *                      https://www.thingiverse.com/thing:1278865
 */

//===========================================================================
//========================== DELTA / SCARA / TPARA ==========================
//===========================================================================
//
// Download configurations from the link above and customize for your machine.
// Examples are located in config/examples/delta, .../SCARA, and .../TPARA.
//
//===========================================================================

[... 2867 lines cut ...]
  #define PRINTER_EVENT_LEDS
#endif

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Set to 0 to turn off servo support.
 */
//#define NUM_SERVOS 3 // Note: Servo index starts with 0 for M280-M282 commands

// (ms) Delay before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Edit servo angles with M281 and save to EEPROM with M500
//#define EDITABLE_SERVO_ANGLES

// Disable servo with M282 to reduce power consumption, noise, and heat when not in use
//#define SERVO_DETACH_GCODE

Uka4me, 2022-12-14:
Thanks for the config. I did not find any differences in it that would solve the problem. I also tried to reinstall the firmware from the repository using `pio run --target erase`, but it still does not work. Maybe someone has some ideas about this?

carlosm180, 2022-12-15:
...try to use menu "reset eeprom"?

flokcer, 2022-12-27:
Same problem. Kura cannot send file to card. Someone was able to solve this? 
[image]
[image]
[image]

Uka4me, 2023-02-14:
I finally got to the printer. I tried resetting the EEPROM with the MKSLaserTool utility, but in the end it did not work.
At the moment I'm trying to debug the write and for now I'm here: Marlin\src\sd\Sd2Card.cpp::DiskIODriver_SPI_SD::writeBlock.

Can someone tell me, I have a sd card with the following characteristics:
Smartbuy
Speed Class: Class 10, UHS Class 1
Maximum write speed: 20 MB/s
Maximum read speed: 70 MB/s

Perhaps others are required, share your characteristics.