makerbase-mks/MKS-TinyBee #76: Filament Runout Sensor is not working on MKS Tinybee and H43 Display

Moditek, 2022-12-11:
I am using "MKS Tinybee marlin" firmware. I could make everything work except "FILAMENT RUNOUT SENSOR" 
I pluged the cable to "MT det" Pin.
I changed parameter that showed me on "MKS Tiny Bee" user manual pdf for configuration.h and adv.h

also I changed pins>pins_MKS_TINYBEE.h.

//
#define X_STOP_PIN                            33
#define Y_STOP_PIN                            32
#define Z_STOP_PIN                            22
#define FIL_RUNOUT_PIN                        35

Please Help

MKS-hosiey, 2022-12-29:
Did you enable the run_out sensor in the configuration file?

Moditek, 2022-12-29:
Yes I did. But When I fill it, All tempature are going fall down. When I try to start printing, Error Message appear. “ Min Temp Foult” something like that.

Windows için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

Kimden: ***@***.***>
Gönderilme: 29 Aralık 2022 Perşembe 05:29
Kime: ***@***.***>
Bilgi: ***@***.***>; ***@***.***>
Konu: Re: [makerbase-mks/MKS-TinyBee] Filament Runout Sensor is not working on MKS Tinybee and H43 Display (Issue #76)

Did you enable the run_out sensor in the configuration file?

Marcel67921, 2023-05-05:
do you have the wires of the runout sensor connector connected to the board well? I had to switch them according to the plate.
- red wire +
- black wire GND
- white wire S

sherwinrastegar, 2023-09-11:
Hello, I have the same problem, and when I compile, it says that the board does not support this display. can you help me ?

Marcel67921, 2023-09-12:
Hello,

check the wiring . + to +, Gnd to Gnd and Signal to Signal. Runout sensors 
differ in the connection of the connector.

Best regards.

---------- Původní e-mail ----------

Komu: makerbase-mks/MKS-TinyBee ***@***.***>
Kopie: Marcel67921 ***@***.***>, Comment ***@***.***
github.com>
Datum: 11. 9. 2023 20:58:23

(https://github.com/makerbase-mks/MKS-TinyBee/issues/76#issuecomment-1714419828)
"

Marcel67921, 2023-09-13:
---------- Původní e-mail ----------

(https://github.com/makerbase-mks/MKS-TinyBee/issues/76#issuecomment-1714419828)
"

Marcel67921, 2023-09-13:
[image]

Marcel67921, 2023-09-13:
S  GND  +5V

Marcel67921, 2023-09-13:
[image]

Marcel67921, 2023-09-13:
[image]