makerbase-mks/MKS-TinyBee #78: M900 not found / Linear Advance not operational

mashuptwice, 2022-12-31:
I'm trying to setup Linear Advance / Pressure Advance. 

M900 is not found and the linear advance calibration pattern lines look all the same. 

I'm trying to find out if the K-Factor settings in Configuration.h are making any difference at all at least, will report back. Even if they do, calibration will be a pain in the A without M900.

mashuptwice, 2023-01-13:
There is now an open upstream Issue: https://github.com/MarlinFirmware/Marlin/issues/25160#issue-1513116272

mashuptwice, 2023-01-14:
Just updated to 2.1.2 and it seems that linear advance is not yet possible with I2S, as the Tinybee uses. I had so many problems with that board, and seeing that MKS doesn't seem to maintain this repository doesn't give me any hope for their producs. I'll ditch the board for a BTT Octopus, they at least merge pull requests.

matthias882, 2023-02-25:
I have the same Problem with A4988 Stepper-Drivers. Linear advance not operational, M900 is respondet with "unknown command"

vfbank, 2023-06-02:
MKS-TinyBee mainboard utilizes the ESP32 chipset. Fundamentally, this chipset communicates through an audio interface called I2C with the Arduino core. The issue lies in the fact that the library for ESP32's Arduino core is still in an 'experimental' beta service. Due to functional compatibility, the Marlin firmware cannot transmit the modified signals to the stepper motor, and as a result, all operations that modulate the signal, including babystepping and the interrupt function, do not work.

Regrettably, the ESP32 board cannot send signals via external references due to its limited communication lines. Unless the Marlin firmware abandons the Arduino core and is written in 32-bit code, it seems very unlikely to support functions like Linear Advance or Input Shaping. That's why this board is inexpensive. If all the aforementioned features were to function, the price of this board would likely increase substantially.