makerbase-mks/MKS-TinyBee #80: [SOLVED] Wrong Temperatures showing in Marlin 2.1.2

MacBravO, 2023-01-21:
I had recurrent problems with inaccurate temperature measurements, below 8~10º of the standard ambient measurement, on both thermistors in marlin version 2.1.2 (official/non-mks repository).

Knowing from the Changelog that Marlin supposedly fixed bugs detailed in these threads:
• https://github.com/MarlinFirmware/Marlin/issues/24142
•https://github.com/makerbase-mks/MKS-TinyBee/issues/7
I noticed that the error was not completely resolved. In the Hal.cpp file ( Marlin-2.1.2\Marlin\src\HAL\ESP32) at line 265 there is a syntax error:

```
#ifndef ADC_REFERENCE_VOLTAGE
   #define ADC_REFERENCE_VOLTAGE 3.3
#endif
```

The " ADC_REFERENCE_VOLTAGE " of the hal.cpp file is declared as 3.3 ( volts), which ends up compromising the " ADC_REFERENCE_VOLTAGE " instruction contained in the pins_MKS_TINYBEE.h file ( Marlin-2.1.2\Marlin\src\pins\esp32), in the line 116:

```
// ADC Reference Voltage
//
#define ADC_REFERENCE_VOLTAGE 2.565 // 2.5V reference VDDA
```

Thus causing a redundancy error (?).

The solution that worked for me was to literally delete 3.3 after the instruction " #define ADC_REFERENCE_VOLTAGE 3.3", in the hal.cpp file. Getting like this:

```
#ifndef ADC_REFERENCE_VOLTAGE
   #define ADC_REFERENCE_VOLTAGE
#endif
```

I did it as a personal whim, also modifying the ADC_REFERENCE_VOLTAGE 2,565 of the hal.cpp file to 2,600, which gave me more precision (variation of 1 to 2 degrees celsius), getting like this:

```
// ADC Reference Voltage
//
#define ADC_REFERENCE_VOLTAGE 2.600 // 2.5V reference VDDA
```

it is. I hope I can help someone, just as I would have liked to have been helped, but I had to find my way on my own.

meetfinger, 2023-02-14:
Thanks for the fix in advance!

Do you have a complete source to compile marlin 2.1.2 for the tinybee? I tried to make it work in vscode from the official build today, but I always get stuck at the display part. I only get white dots on a pulsating 12864v3. Dunno if it’s because only the display part is faulty or if my dozens of other modifications broke the whole build.

3DSmitty, 2023-05-30:
@MacBravO I had to adjust my ADC_REFERENCE_VOLTAGE as well. I wonder if it depends on the type of thermister used. I use a type "13" and the default high temps were off by like 20C the "3.3" is what worked for me and got me within +-1 degrees.