makerbase-mks/MKS-TinyBee #81: TINYBEE Issue with temperature over 283 degres in Marlin 2.1.2

JumboNicoStras, 2023-02-02:
Hello,

I have a Tinybee board with Marlin 2.1.2.
The Maxtemp is at 320 degres.
The Hotend_Overshoot is at 15 degres.

Issue: If i ask for a temperature at 290 degres, the screen froze at 283 degres and the nozzle continue to heat up without any regulations and thermal security. (reading on the screen stay at 283 degres but the nozzel can be at 350 or more !!)

Someone has the same problem a me ?

Note: If i use the same Marlin on the S-Gel L V2 Board (only changing the board declaration) , there is no issue.

Can you help me, please ?

flishy2532, 2023-03-04:
use marlin bugfix 2.1.x 
 [image]

 [image]

JumboNicoStras, 2023-03-06:
Super, i will try .
What are exatly the differences between the Current Marlin Release , Marlin "Nightly" Source and the Marlin Patched Source ?
Thank you

flishy2532, 2023-03-06:
am use Nightly am try current marlin it have bug

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>