makerbase-mks/MKS-TinyBee #83: Extruder as z motor

enddDD, 2023-03-14:
Hi, can i use an extruder motor, like e0 or e1 as z to have 3 z? Also, wil this work with klipper or it is marlin only? Thanks!

MKS-hosiey, 2023-03-23:
This motherboard currently supports marlin firmware, and the motherboard supports up to 2Z

enddDD, 2023-03-23:
> This motherboard currently supports marlin firmware, and the motherboard supports up to 2Z

I cant replace e0 or e1 to make a z with it ? I have the board and have 3 z, i see that the board has only 2, but has also 2 extruders-i need only one. Thanks.

Eng-Miro, 2023-08-10:
Hello,
I came by your problem
i was checking the pins configuration in the MKS tinybee
you have already Z2 configured in the same position of E1 (the second extruder)
"#define E1_STEP_PIN                          141
#define E1_DIR_PIN                           142
#define E1_ENABLE_PIN                        140

#define Z2_STEP_PIN                          141
#define Z2_DIR_PIN                           142
#define Z2_ENABLE_PIN                        140"

theoretically, you just need to enable Z2 in config.h
then connect two motors in the green slots (both will be drived by Z stepper driver)
Then connect the third motor in the purble slot (will be drived by Z2 Stepper driver)

Hopes this will work for you