makerbase-mks/MKS-TinyBee #84: MKS TinyBEE TMC2209x4 AnyCubic Kossel Linear Plus Last Fix LayerShift

flishy2532, 2023-04-02:
My Printer Spec AnyCubic Kossel Linear Plus 2023

MKS-TINYBEE Marlin BugFix2.1.x  LayerShift Fix 
-TMC2209 x4 
-MKS MINI v3.0 SDCARD SUPPORT 
-RBG LED 
-WIFI SUPPORT 
-FIXBUG LayerShift 
-Smooth Print 

FIRMWARE EXE FILE

https://drive.google.com/file/d/1lP4uoKXvk038l4Codv4hUs6DFZCI3kRr/view?usp=share_link

FIRMWARE RAR FILE

https://drive.google.com/file/d/1NPF3403I5VFHS4_3TrFFqpdcggzXvwOP/view?usp=share_link

flishy2532, 2023-05-18:
> Could u advise on how u fixed the layershift problem? or what configs u changed

//define print counter

enddDD, 2023-05-27:
This printercounter caused to layersift on my printer too. Something is wrong with it. Disabled it and now it seams to work. Thanks for the info

flishy2532, 2023-05-27:
no problem hope fun on print

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>