makerbase-mks/MKS-TinyBee #85: MKS tinybee web UI

irkzome117, 2023-04-03:
I uploaded a wrong index.html.gz file and now i don't know how to change it and upload de right one.

When i connect to Marlin wifi it open the portal, but still on loading page.

Marcel67921, 2023-04-10:
I have the same problem.I uploaded a wrong index.html.gz file and now i don't know how to change it and upload de right one.

When i connect to Marlin wifi it open the portal, but still on loading page. Please advise me how to solve this problem.

PanosPetrou, 2023-04-25:
Go to http://0.0.0.0?forcefallback=yes

Replace 0.0.0.0 with the actual ip address.
Delete the index.html.gz file and upload the correct one.

Marcel67921, 2023-05-02:
I solved the problem by using another PC to update. I had to erase the flash on the board using Visual Studio Code. Then I updated the board.

Marcel67921, 2023-05-02:
[https://github.com/makerbase-mks/MKS-TinyBee/issues/5]