makerbase-mks/MKS-TinyBee #87: Web UI/MarlinESP access point configuration reset

GameFan01, 2023-04-28:
Hi,
Is there a way I can reset the access point configuration to factory default?
I think I've accidentally entered an invalid IP address and now the web UI refuses connection every time.

Thank you.

Marcel67921, 2023-05-02:
I solved the problem by using another PC to update. I had to erase the flash on the board using Visual Studio Code. Then I updated the board.

GameFan01, 2023-05-02:
> I solved the problem by using another PC to update. I had to erase the flash on the board using Visual Studio Code. Then I updated the board.

How did you update the board? By reflashing the firmware?

Marcel67921, 2023-05-03:
> Komu: makerbase-mks/MKS-TinyBee ***@***.***>
> github.com>
> configuration reset (Issue #87)
> "
>
> "
> Problém jsem vyřešil pomocí jiného PC k aktualizaci. Musel jsem vymazat
> flash na desce pomocí Visual Studio Code. Pak jsem aktualizoval desku.
> "
> Jak jsi aktualizoval desku? Přeflashováním firmwaru?
>
> Odpovězte přímo na tento e-mail, zobrazte jej na GitHubu
> (https://github.com/makerbase-mks/MKS-TinyBee/issues/87#issuecomment-1532193567)
> nebo se odhlaste z odběru
> "

Yes,
using Visual Studio Code and its plugin Platform IO

---------- Původní e-mail ----------

GameFan01, 2023-05-03:
Just tried using the Platform IO "clean" option in VS code. Then I uploaded the firmware provided here by makerbase without modifying anything, and before the upload I used the "forget" option on the Marlin_ESP access point. Nothing changed, it still refuses connections. 
Here's my output from compiling and uploading the firmware if you can see something wrong in it:
[Compile output.txt](https://github.com/makerbase-mks/MKS-TinyBee/files/11385041/Compile.output.txt)

Marcel67921, 2023-05-04:
have you tried initializing the eeprom in the menu settings on the lcd 
display after uploading the firmware?

---------- Původní e-mail ----------

Datum: 3. 5. 2023 18:57:13

(https://github.com/makerbase-mks/MKS-TinyBee/files/11385041/Compile.output.txt)

(https://github.com/makerbase-mks/MKS-TinyBee/issues/87#issuecomment-1533384830)
"

GameFan01, 2023-05-04:
I tried initializing the eeprom now and the result is still the same.

Marcel67921, 2023-05-04:
i used this

I still have the VSC Auto Build Marlin plugin installed

Project tasks in Platform IO

1. Erase Flash

2. Build

3. Upload

4. Eeprom initialization

check the Configuration.h and Configuration_adv.h settings according to the 
mks_tinybee board manual   https://github.com/makerbase-mks/MKS-TinyBee
(https://github.com/makerbase-mks/MKS-TinyBee)

index.html.gz and files I used these https://github.com/makerbase-mks/MKS-
TinyBee/issues/5(https://github.com/makerbase-mks/MKS-TinyBee/issues/5)

Hopefully it will help you

---------- Původní e-mail ----------

Datum: 4. 5. 2023 15:11:12

(https://github.com/makerbase-mks/MKS-TinyBee/issues/87#issuecomment-1534756640)
"

GameFan01, 2023-05-04:
OH MY GOD IT WORKS!
Thank you so much. I thought the board was bricked for good.

closed 2023-05-04 by GameFan01

Marcel67921, 2023-05-05:
I'm glad it worked out for you

metkosamra, 2024-01-12:
> أنا يتسعدني أن أخصص لك

Thank you very much, I had the same problem and you solved it