makerbase-mks/MKS-TinyBee #88: MKS tinybee board

ahmedhassan3298, 2023-05-03:
Hi all,
my MKS tinybee board give error in temperature when i connect end stop for y-axis
[image]
[image]

enddDD, 2023-05-03:
https://github.com/makerbase-mks/MKS-TinyBee/issues/36
Check this, maybe it helps you. Try using only 2 pins for the endstops

ahmedhassan3298, 2023-05-03:
Thanks for your help
I Try 2 pins (S,G) for the end stop The end stop dont stop the motor

enddDD, 2023-05-03:
Is it NC (normally closed) the switch? If the switch has 3 pins u should tie the two from each side and leave the middle pin unconnected.

ahmedhassan3298, 2023-05-03:
[image]

i use this type of endstop i disconnect the red wire

closed 2023-05-04 by ahmedhassan3298