makerbase-mks/MKS-TinyBee #91: Nozzle heater intermitent error

meipsum, 2023-05-25:
The nozzle heater suddenly stopped working; I turned the printer on one morning, and it wouldn't heat. Thermal runout error. I ordered it to heat both the bed and the nozzle and measured the voltage at the motherboard connections; bed OK (a tad less than 12v), nozzle zero. I changed the pins in Marlin so that the nozzle would use the connection for a secondary extruder I don't have, and it would not work either: zero volts at both 145 and 146 (regular and secondary). 
Bought another board, and the very next day, after a few test prints, the same issue: zero volts when told to heat the nozzle. Turned it off and on a few times, heated the nozzle connecting it straight to the power source in order to take the filament out, and all of a sudden it started working again.
Now I fear that at any moment it will stop heating the nozzle again.

PanosPetrou, 2023-05-26:
You have probably a poor cable connection somewhere, or a short. Keep in mind the polarity when you connect things. A lot of connectors are having a common ground, make sure your cable connections are not reversed, you might be shorting something.

meipsum, 2023-05-26:
> You have probably a poor cable connection somewhere, or a short. Keep in mind the polarity when you connect things. A lot of connectors are having a common ground, make sure your cable connections are not reversed, you might be shorting something.

I just found out that every single time it refuses to heat the nozzle (and I get a beeping error) it works OK if I first heat the nozzle a bit by connecting it to 12V first. It's an Anet A8 thermistor, set up in Marlin as "5 : 100kΩ ATC Semitec 104GT-2/104NT-4-R025H42G - Used in ParCan, J-Head, and E3D, SliceEngineering 300°C" according to the manufacturer's instructions. It always starts as if it was 0oC (room temp here is around 20oC), but in higher temperatures, it is quite accurate. The thermal overrun error is not due to taking time to heat; it just doesn't start sending 12V to the hotend if it doesn't feel like it. But if I get it a bit hot before, it works perfectly. Seems to be some weird bug. Perhaps it reads the heating element's resistance and if it is not within some range it doesn't heat? I really don't know. I could try adding a resistor to see if it's the case. But it's certainly not a loose cable or anything that simple.

PanosPetrou, 2023-05-27:
Were you using the same power supply on both boards? I'm starting to think something is wrong with your power supply. Furthermore, you have a 12V system which will require more amps to function, compared to a 24V system. Perhaps the board (its mosfets) can't handle delivering that much amperage, but it's just a theory.

In my case, I'm using a 24V 480W power supply, the same board with two extruders, two ATC Semitec 104GT-2 thermistors and 50W heaters and it works reliably. The only issue I had, was that I initially connected one of the thermistors with wrong polarity which caused the other thermistor to get a false temperature reading. I changed the polarity and everything worked fine afterwards.

meipsum, 2023-05-28:
> Were you using the same power supply on both boards? I'm starting to think something is wrong with your power supply. Furthermore, you have a 12V system which will require more amps to function, [...]

Well, a lack of amps would make it hard(er) to heat, period. I'll see if I have a 24V power supply somewhere, but I don't think it will solve the issue, as it does heat **if** the temperature is above (the false reading of) zero celsius (and sometimes when it is not). Curiously, I found out that if I connect the minus wire of the heater with the minus wire from the source it works just as well as if I connect the plus wire too. I left it connected for a while to see where it would go and it reached 265oC and kept around that temperature. 

I'll also try reverting the hotend thermistor, as the issue does not appear in the bed heater and it is highly improbable that the underlying logical steps are different. Perhaps I inadvertently miswired it. Thanks for the tip.

PanosPetrou, 2023-05-28:
If you are going to use a 24V power supply, make sure your heaters are rated for 24V too. If they are rated for 12V you will need to replace them, otherwise they are going to burn out.