makerbase-mks/MKS-TinyBee #93: This board is waste of money

enddDD, 2023-06-04:
No big settings seames to work with this. 
Input shaping-no
Lin advanced-no
Not even babysteps....

It says in the adds that it works with marlin, well...it doesnt!

PanosPetrou, 2023-06-06:
This board is using an ESP32 processor which is a newer thing in 3D printing, compared to other boards with STM processors. It's understandable some features are not supported yet because of the different architecture.
If you haven't done your research and purchased this board blindly, it's not the board's fault.

enddDD, 2023-06-07:
> 

Well, i dont really care what processor is using, it says it supports marlin, well at least 3 main functions it doesnt. Is like i sell u a car with square weels, it runs but you wil hop like crazy in it. If it doesnt support those firmware functions is pretty useless, just to move 5 motors arround and heat a hotend-ok- this even ramps1.4 can do. Dont advertise something that does not work. Simple.

PanosPetrou, 2023-06-07:
Your analogies are flawed, it boots and runs Marlin, so it's advertised correctly. However none of the boards are required to support all features. Furthermore, Marlin is an open source project, none of the developers are obligated to provide 100% compatibility for every single board. And as I said, it's a newer architecture which is not fully compatible will all features (yet).

vfbank, 2023-06-23:
If you look at the main header.cpp of the ESP32 processor, you can see that it lacks an I2C stream call function. This means that this chipset cannot utilize Marlin's I2C stream bus function.

The ESP32 chipset was developed as a replacement for Arduino during the supply disruption of the STM32 chipset, specifically for 3D printer applications. However, this chipset was originally made for IoT device communication. Now, as the STM32 chipset is no longer facing supply issues, we can essentially consider that the development of motherboards made with the ESP32 chipset has been discontinued.

vfbank, 2023-06-23:
Looking at the main header of the ESP32 processor, you can see that it lacks an I2C stream call function. This means that this chipset cannot utilize Marlin's I2C stream bus functionality. Therefore, all functions in the original source that modulate signals, such as babystep, linear advance, and input shaping, cannot be used with this chipset.

The ESP32 chipset was hurriedly developed as a replacement for Arduino during the period of supply disruption of the STM32 chipset, specifically for use in 3D printers. However, this chipset was originally manufactured for communication in IoT devices, and due to communication protocol issues, it's not suitable for 3D printers. Now that the STM32 chipset is no longer experiencing supply difficulties, the development of Arduino-compatible mainboards using the ESP32 chipset has essentially been discontinued. This is why MKS no longer supports this chipset.

Production has already stopped since last year. MKS is under serious stock pressure and is currently supplying below manufacturing cost on platforms like AliExpress. For reference, the initial launch price of this mainboard was $34, but now it's being sold in the range of $19 to $23.

FrankHovis, 2024-02-01:
> This is why MKS no longer supports this chipset.

MKS never support any of their products beyond "Here is something that half works".