makerbase-mks/MKS-TinyBee #98: MKS tinybee V 1.0 does not power on

ryan152159, 2023-08-09:
Just got a tiny bee V1.0 for 3d printing, When using a USB cable to connect to my computer, the LED does not light up (which I assume it should) although I have changed it to USB power on. Is there an issue with the board itself?

*Disclaimer* I am very new here and really know almost nothing so thanks for helping me out.

SubJORJ, 2023-08-22:
Hello.
The board must be powered by USB. In this case, the USBPWR jumper must be set to the ON position (right contact). I also advise you to disconnect all external devices, stepper motor drivers, display, etc. from the board.

If the board is not working, then your USB port may not be supplying the required current, or the USB cable is too thin and long.

PanosPetrou, 2023-12-10:
In my case I had to set the jumper to USBPWR and also connect it to a USB powered hub, I guess it needed more power. Keep that in mind.