makerbase-mks/MKS-TinyBee #99: MKS Tinybee reads wrong temperature

ducntq, 2023-08-21:
My MKS Tinybee suddenly reads temperature of hotend at 133°C and bed at 133°C, even though room temperature is at constantly 26°C (AC room). I was just using the board normally, until I turnt the board on this morning. I tried replacing the thermistors for both hotend & bed, but the board reads incorrect value. I tried unplugged both thermistors, the board reads 130°C.

The firmware I'm using here (although I doubt it was a firmware issue): https://github.com/ducntq/mks-tinybee-marlin

SubJORJ, 2023-08-23:
Check voltage at Zener diode D7, should be approximately 2.5V. Maybe your zener diode has shorted out, or the large 30 ohm resistor has burned out or is badly soldered.

JazzinSA, 2023-10-05:
I use a large number of these motherboards. I have found that the problem is not a marlin issue. Some motherboards is just way off. From a batch of 10 motherboards I will find 2 or 3 that the temperature is off sometimes as much as 20 degrees. If I were better at writing marlin code I would have written a calibration procedure but unfortunately my marlin skills is just not there yet.