makerbase-mks/MKS-TinyBee #106: Bybystepping in real time.

Marcel67921, 2023-12-03:
Could someone please advise me to make bybystepping work for me when printing the first layer. In real time.

PanosPetrou, 2023-12-05:
Babystepping isn't currently supported on ESP32 boards, such as this one. Perhaps it will be supported in the future.

Marcel67921, 2023-12-05:
Thank you, hopefully it will be possible in the future.
---------- Původní e-mail ----------

Komu: makerbase-mks/MKS-TinyBee ***@***.***>
Kopie: Marcel67921 ***@***.***>, Author ***@***.***
com>
Datum: 5. 12. 2023 3:39:42
106)
"

(https://github.com/makerbase-mks/MKS-TinyBee/issues/106#issuecomment-1839901599)
"

vfbank, 2024-03-05:
Unfortunately marlin does not support a single I2C stream for AVR motherboard support. 

Supporting Babystep on the ESP32 would require giving up AVR compatibility, but that is not the goal of the marlin project. Use klipper instead of marlin