makerbase-mks/MKS-TinyBee #107: Schematic for MKS tinybee

Elentriekerke, 2023-12-25:
Hi for some reason fan2 is suddenly at 100% duty cycle all the time, even after rewrite or reset. So I was wondering if there are schematics available for this board in order to find the issue. 

I can see a red LED lighting up for one of the fans, maybe there's some G code to reset the pwm driver.

Thanks for the help in advance 🙏

thmalmeida, 2023-12-26:
Even the microcontroller (ESP32) has pwm on many pins, tinybee board does not offer direct connection to drives that controls fans or heaters. Perhaps due the few number of pins available. And it uses three 8 bit shift registers to control those parts using I2S peripheral. So, a pwm output on fans or heaters is not possible by hardware. The fan speed can be controlled by some more code to construct a pwm signal by software with low frequency.

Are you using Marlin firmware? What is the firmware version?

Elentriekerke, 2023-12-26:
It worked before as it is now, so i think something went wrong causing this issue... As far as I found schematics from previous versions the fans are controlled by a separate chip, which gets its data from the esp32.

But however there's a red led close to the fan output which is now red lit. 

I use marlin 2.1.2...

Marcel67921, 2024-01-15:
Could you please provide the configuration on the TinyBee motherboard for the Marlin-2.1.2.1 . I would add the data for setting the board size, limits and speeds. I'm not an expert in setting up the configuration. Thank you.

viernullvier, 2024-02-12:
@Elentriekerke Did you check if your fan's `-` connection might be shorted against `GND`? I'm asking because the fan's `+` pin is always connected to V<sub>in</sub> while the `-` pin is the one that's getting switched (between not connected / off and `GND` / on). Shorting `-` to `GND` will bypass the microcontroller / PWM signal, power the fan connector and light up the LED permanently.

danvie4, 2024-06-11:
@Elentriekerke Have you looked at the board to see if there's any physical damage (blown mosfet, other components that look burnt...)?