makerbase-mks/MKS-TinyBee #108: end stop temperature error

scymz, 2023-12-28:
When my x and y optical endstops are closed, the temperatures appear as err. It has been said in some places that it is related to 5v. Has anyone encountered this situation and been able to solve it?

PanosPetrou, 2024-03-01:
Most likely your endstops are wired with a wrong pin order and when they engage, they short circuit the 5V rail. Check your wiring and pinout.

scymz, 2024-04-30:
> Most likely your endstops are wired with a wrong pin order and when they engage, they short circuit the 5V rail. Check your wiring and pinout.

My cable connections are correct, I make connections according to the order on the board. When I disconnect the 5v connection on the mechanical endstop, it works properly, but it does not work on the optical endstop, it always appears in the triggered position.